pulp-platform/pulpissimo #274: Mismatch between simulation and FPGA emulation behavior

yttuncel, 2021-09-29:
Hi all,

The HWPE behavior between simulation and FPGA is different. I've tried it with pulp_soc v2.1.0, v3.0.0 and v3.0.1. They all show the same problematic behavior so far.

**INFO:** I'm using the default mac engine hwpe. In the hwme.c example in pulp-rt-examples, I made couple small changes. I use the simple multiplication mode, and load 0 to 127 to *a and *b. I write 5s from *d to *(d+127) in the beginning to verify that they are overwritten. I expect to see 0 1 4 9 16 ... in *d after the hwpe completes operation. I also print the contents of the HWPE registers before and after configuration as sanity check. Now the results:

1. **SIMULATION**: The simulation behaves as expected (see below). The event is generated, and I can see the correct values at the location pointed by *d. The register contents are also correct, e.g. a_addr is set to 0x1c010000. (In hwme.c, I had to move hwme_trigger_job(); between the _setEvent() and _wait_event() function calls for the event to work properly, as pointed out in earlier issues).

2. **FPGA**: I synthesize the design for ZCU102. I set USE_HWPE=1 in xilinx_pulpissimo.v. I build the application with 'make all io=uart' and with -O0 flag. I add 
```'
int __rt_fpga_fc_frequency =     20000000; // e.g. 20000000 for 20MHz;
int __rt_fpga_periph_frequency = 10000000; // e.g. 10000000 for 10MHz;
unsigned int __rt_iodev_uart_baudrate = 115200;
```
at the top of hwme.c. When I run this application on the FPGA, the output I get from UART on the screen terminal is completely wrong (see below again). It looks like the HWPE registers are not being configured correctly, or rather the last register write is being spread to all registers.  I'm also checking the contents of the registers in gdb using examine, and they match with the printf outputs.

**However, how can the event be generated if that's the case?**
**But if the event is generated, then why am I not seeing expected output at \*d?**

I appreciate any pointers at this point. _**Am I missing some steps?**_ 
I've spent quite a lot of time on this and honestly am out of ideas.

Below are the two outputs. I can provide the hwme.c if requested, it's pretty much the original code with small changes here and there.
```
**Simulation:**
[STDOUT-CL31_PE0] d[0]=5
[STDOUT-CL31_PE0] d[1]=5
[STDOUT-CL31_PE0] d[2]=5
[STDOUT-CL31_PE0] d[3]=5
[STDOUT-CL31_PE0] d[4]=5
[STDOUT-CL31_PE0] d[5]=5
[STDOUT-CL31_PE0] d[6]=5
[STDOUT-CL31_PE0] d[7]=5
[STDOUT-CL31_PE0] d[8]=5
[STDOUT-CL31_PE0] d[9]=5
[STDOUT-CL31_PE0] Register contents before config:
[STDOUT-CL31_PE0] a addr:=0
[STDOUT-CL31_PE0] b addr:=0
[STDOUT-CL31_PE0] c addr:=0
[STDOUT-CL31_PE0] d addr:=0
[STDOUT-CL31_PE0] nb iter:=0
[STDOUT-CL31_PE0] len iter:=0
[STDOUT-CL31_PE0] Register contents after config:
[STDOUT-CL31_PE0] a addr:=1c010000
[STDOUT-CL31_PE0] b addr:=1c010200
[STDOUT-CL31_PE0] c addr:=1c010400
[STDOUT-CL31_PE0] d addr:=1c010600
[STDOUT-CL31_PE0] nb iter:=1
[STDOUT-CL31_PE0] len iter:=7f
[STDOUT-CL31_PE0] Operation completed:
[STDOUT-CL31_PE0] d[0]=0
[STDOUT-CL31_PE0] d[1]=1
[STDOUT-CL31_PE0] d[2]=4
[STDOUT-CL31_PE0] d[3]=9
[STDOUT-CL31_PE0] d[4]=16
[STDOUT-CL31_PE0] d[5]=25
[STDOUT-CL31_PE0] d[6]=36
[STDOUT-CL31_PE0] d[7]=49
[STDOUT-CL31_PE0] d[8]=64
[STDOUT-CL31_PE0] d[9]=81
```
_______________________________
```
**FPGA:**
d[0]=5
d[1]=5
d[2]=5
d[3]=5
d[4]=5
d[5]=5
d[6]=5
d[7]=5
d[8]=5
d[9]=5
Register contents before config:
a addr:=11a13c05
b addr:=11a13c05
c addr:=11a13c05
d addr:=11a13c05
nb iter:=11a13c05
len iter:=11a13c05
Register contents after config:
a addr:=7f
b addr:=7f
c addr:=7f
d addr:=7f
nb iter:=7f
len iter:=7f
Operation completed:
d[0]=5
d[1]=5
d[2]=5
d[3]=5
d[4]=5
d[5]=5
d[6]=5
d[7]=5
d[8]=5
d[9]=5
```

meggiman, 2021-09-29:
Hi @yttuncel,

We currently don't have the time to debug this issue ourselves. The way I would approach this problem would be to use XILINX ILA cores in Vivado to make the signals of interest within the HWPE accelerator observable in real-time. You can then compare these signals with the RTL simulated counterparts and should be able to get a clue where and why the two platforms diverge. To be honest I cannot think of a reason why the two platforms should behave differently so digging into the internal signals using the in-circuit debugging capabilities of Vivado is the methodology I would choose to investigate the issue.

Best,
Manuel

yttuncel, 2021-09-30:
@meggiman : Thank you for your suggestion. We connected the required signals to an ILA core and generated a bitstream.
However, we are not able to connect to both the JTAG on J83 in ZCU102 FPGA (for ILA debug) and the JTAG-HS2 for the RISCV core for OpenOCD simultaneously.

The libusb driver errors out stating that the libusb_claim_interface() failed with LIBUSB_ERROR_BUSY. We did some debugging and noticed that both use the same Digilent controller with the same serial numbers. So, we are unable to proceed with ILA.

**Is there a way to connect to both the on-board JTAG interface and the JTAG-HS2 interface simultaneously such that we can execute applications on the RISCV core and observe the waveforms on the ILA in Vivado?**

yttuncel, 2021-09-30:
Hi Manuel,

> We currently don't have the time to debug this issue ourselves.

I just wanted to reiterate that we are trying to run the HWPE example without making any changes to it.
So, to confirm, **does the HWPE example work out of the box on ZCU102?** 
Has anyone on your end recently tried it?

Is there a specific Vivado version that we should use to generate bitstream?

Best,
Yigit

sevjaeg, 2022-10-27:
I am experiencing the same issue (with pulpissimo v7.0.0, pulp_soc v3.0.0, hwpe-mac-engine v1.3.3, hwpe-ctrl v1.6.1, and the template HWCE code). Everything worked out nicely in the RTL simulation, however no data is written to the output address after FPGA synthesis for the Genesys2 board (as well as in a post-synthesis simulation after a quick ASIC synthesis run). Still, the execution terminates as the event is triggered. Thus, I assumed there was some issue with the output addresses.
As @yttuncel I got incorrect values after reading out the configuration registers. Then, I tried to hardwire the signals set via the registers in `mac_ctrl.sv` and `mac_fsm.sv`. This worked fine both for FPGA and ASIC. Thus, I would assume that there is some issue with the latch-based register file. However, this is definitely no satisfying solution...

zealshah29, 2024-04-23:
> Hi all,
>
> The HWPE behavior between simulation and FPGA is different. I've tried it with pulp_soc v2.1.0, v3.0.0 and v3.0.1. They all show the same problematic behavior so far.
>
> **INFO:** I'm using [...]

Hi @yttuncel  can you share the hwme.c code file that you changed in pulp-rt-examples? Also apart from that, did you make changes in any other files?

zealshah29, 2024-04-25:
Hi @sevjaeg ,
So I am trying to run the example hwme.c code and I am currently stuck at the line `--rt_periph_wait_event`. I made the changes as specified in this issue, yet I am not able to get the output. If you were able to successfully run hwme.c code on fpga, could you guide me on what all changes did you make to all the files? If you have a reference report for the same, that would help me out too. 
Thanking you. 
Regards, 
Zeal

zealshah29, 2024-04-30:
Hi @sevjaeg 

> I am experiencing the same issue (with pulpissimo v7.0.0, pulp_soc v3.0.0, hwpe-mac-engine v1.3.3, hwpe-ctrl v1.6.1, and the template HWCE code). Everything worked out nicely in the RTL simulation, however no data is written to the output address after FPGA synthesis for the Genesys2 board (as well as in a post-synthesis simulation after a quick ASIC synthesis run). Still, the execution terminates as the event is triggered. Thus, I assume there is some issue with the output addresses.
> 
> As @yttuncel I got incorrect values after reading out the configuration registers. Then, I tried to hardwire the signals set via the registers in `mac_ctrl.sv` and `mac_fsm.sv`. This worked fine both for FPGA and ASIC. However, this is definitely no satisfying solution.
> 
> Thus, I would assume that there is some issue with the latch-based register file...
I am also stuck over here. I checked your hwpe-aes repo, and saw the changes you made in mac_ctrl and mac_fsm, I also made the similar changes and tried running the code. But d's value is not getting updated. I will share the changes that I made in each file over here for your reference. 

Changes in the mac_ctrl file:
```
 /* Direct register file mappings */
  assign static_reg_nb_iter    = 1;
  assign static_reg_len_iter   = 16;
  assign static_reg_vectstride = 4;
  assign static_reg_onestride  = 4;
.
.
.
  always_comb
  begin

    fsm_ctrl.simple_mul = '0;
    fsm_ctrl.shift      = '0;
    fsm_ctrl.len        = static_reg_len_iter[$clog2(MAC_CNT_LEN):0];
  end
```
Changes in the mac_fsm file:
Added the following line
```
  assign ctrl_uloop_o.ready = 1'b1;
  ```
  and hardcoded the address
  
  ```
      ctrl_streamer_o.a_source_ctrl.addressgen_ctrl.base_addr   = 32'h1c010000 + (flags_uloop_i.offs[MAC_UCODE_A_OFFS]);
      ctrl_streamer_o.b_source_ctrl.addressgen_ctrl.base_addr   = 32'h1c010200 + (flags_uloop_i.offs[MAC_UCODE_B_OFFS]);
      ctrl_streamer_o.c_source_ctrl.addressgen_ctrl.base_addr   = 32'h1c010400 + (flags_uloop_i.offs[MAC_UCODE_C_OFFS]);
      ctrl_streamer_o.d_sink_ctrl.addressgen_ctrl.base_addr   = 32'h1c010600 + (flags_uloop_i.offs[MAC_UCODE_D_OFFS]); 
```

Also I commented the following code in hwme.c code:
```
   // job-dependent registers
    //hwme_a_addr_set((unsigned int) a);
    //hwme_b_addr_set((unsigned int) b);
    //hwme_c_addr_set((unsigned int) c);
    //hwme_d_addr_set((unsigned int) d);
 ```
 
 I have a few questions:
 1. Apart from these, were there any other changes that you made in your code files?
 2. I saw in your aes-demo.c code that you have removed the code where it sets the bytecode, like this in hwpe.c
 ```
    // set up bytecode
    hwme_bytecode_set(HWME_LOOPS1_OFFS,           0x00000000);
    hwme_bytecode_set(HWME_BYTECODE5_LOOPS0_OFFS, 0x00040000);
    hwme_bytecode_set(HWME_BYTECODE4_OFFS,        0x00000000);
    hwme_bytecode_set(HWME_BYTECODE3_OFFS,        0x00000000);
    hwme_bytecode_set(HWME_BYTECODE2_OFFS,        0x00000000);
    hwme_bytecode_set(HWME_BYTECODE1_OFFS,        0x000008cd);
    hwme_bytecode_set(HWME_BYTECODE0_OFFS,        0x11a13c05);
 ```
Did you hardcode its address as well? If yes, where? 
    
3. After I made the changes, I ran the following two commands:

 ```
    cd pulpissimo
    sudo make checkout
 ```
to reflect the changes I made in hwpe-mac-engine ip 
    
and inside pulp-rt-examples/accelerators/hwme folder:
    
 ```
   make clean all io=uart
 ```
   
to recompile the updated hwme.c code
Am I missing any steps over here that I need to execute after manually changing hwpe-mac-engine ip?