pulp-platform/pulpissimo #275: Issue with make build in RTL simulation

Merok, 2021-09-30:
Hello,

I have an error while trying to build the RTL simulation platform.
I have followed the readme and installed all the requirements. 
I work on "Rocky Linux "8.4 (Green Obsidian).
Here is my log: 

make build
cd sim && make all
make[1]: Entering directory '/fol/XX/pulpissimo/sim'
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make[3]: Nothing to be done for 'lib'.
make -C ../rtl/tb/remote_bitbang all
make[2]: Entering directory '/fol/XX/pulpissimo/rtl/tb/remote_bitbang'
cc -MT remote_bitbang.o -MMD -MP -MF ./.d/remote_bitbang.Td -std=gnu11 -fno-strict-aliasing -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-function -Wno-missing-braces -O2 -g -march=native -DENABLE_LOGGING -DNDEBUG -fPIC -I./  \
	-c  remote_bitbang.c -o remote_bitbang.o 
ld -shared -E --exclude-libs ALL -o librbs.so  \
	remote_bitbang.o sim_jtag.o 
make[2]: Leaving directory '/fol/XX/pulpissimo/rtl/tb/remote_bitbang'
/fol/XX/pulpissimo/sim//tcl_files/rtl_vopt.tcl
QuestaSim-64 vopt 2020.4 Compiler 2020.10 Oct 13 2020
** Error (suppressible): (vopt-14401) Option specified: +acc=mnpr. Following acc options are now deprecated 'f' 'm' 'q' 't' 'x' 'y'.
Errors: 1, Warnings: 0
child process exited abnormally
    while executing
"exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp -L /fol/XX/pulpissimo/install/modelsim_libs/common_cells_lib..."
    ("eval" body line 1)
    invoked from within
"eval exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp  $VSIM_IP_LIBS $VSIM_RTL_LIBS -work work "
    invoked from within
"if {[info exists ::env(VSIM_PATH)]} {
    #eval exec >@stdout vopt +acc=mnpr -o vopt_tb tb_pulp -floatparameters+tb_pulp -Ldir $::env(VSIM_PATH)/model..."
    (file "/fol/XX/pulpissimo/sim//tcl_files/rtl_vopt.tcl" line 26)
make[1]: *** [Makefile:43: opt] Error 1
make[1]: Leaving directory '/fol/XX/pulpissimo/sim'
make: *** [Makefile:114: build] Error 2

I am a bit confused on my have such error.
Do you have an idea ?

Thanks a lot for the help.

bluewww, 2021-09-30:
** Error (suppressible): (vopt-14401) Option specified: +acc=mnpr. Following acc options are now deprecated 'f' 'm' 'q' 't' 'x' 'y'. @meggiman

Merok, 2021-10-05:
Hi,

Any help on this issue would be greatly appreciated.

Thanks for the help

meggiman, 2021-10-05:
Hi @Merok 

This is an issue with the now depcrecated vopt acc flags in newer versions of Questasim. Until we remove those flags in a future bugfix release of PULPissimo, you can workaround the issue by changing the file sim/tcl_files/rtl_vopt.tcl manually. Change ... vopt +acc=mnpr ... to ... vopt +acc ... (wihtout the '=mnpr') on line 28 and line 30. This should do the trick.

Merok, 2021-10-05:
Hi,

This solved my issue.
Thanks for the help!

A-Eren, 2023-10-09:
Have the same problem with changed rtl_vopt.tcl 
[image]

A-Eren, 2024-05-09:
Reinstalled the whole envoirement till its works

rtspk, 2024-05-09:
@A-Eren , could you please tell me, which tool version of Mentor Questa SIM did you use to simulate the design?

A-Eren, 2024-05-10:
Intel® Quartus® Prime Standard Edition Design Software Version 21.1.1

rtspk, 2024-05-10:
@A-Eren Sorry that I am bothering you with a lot of questions. I am just trying to narrow down the problem. 

I have an additional question. Is your simulator a free version (i.e., a starter version), or do you have a paid license version?

A-Eren, 2024-05-10:
I used student license

rtspk ***@***.***>, 11 May 2024 Cmt, 01:32 tarihinde şunu
yazdı: