pulp-platform/pulpissimo #289: SDK hello example won't run on FPGA

rijun, 2021-12-01:
Hello,

I'm trying to implement the Pulpissimo on a Nexys 4 DDR, and have some issues running an application on the FPGA.

I followed the instructions in the README to generate the bitstream. For compiling the hello example from *pulp-rt-example*, I used both the SDK with tag v1 as well as the SDK with tag 2019.12.06, and compiled the application with the command `make clean all io=uart platform=fpga`. OpenOCD, also compiled as described in the README, recognizes the RISC-V core and I can upload the application to the core, but afterwards nothing happens.
Setting breakpoints at `_entry` or in `main` does not help as these do not get triggered, and on Ctrl+C the processor is stuck at the address `0x00000000`.

I have tried out the current master branch of Pulpissimo, as well as the releases v7 and v6, and all of these show the same behaviour. But with the Pulpissimo of commit dde07e1everything worked.

Regards,
rijun

benoit1995, 2021-12-07:
@rijun 
I also used Nexys4DDR and had the same problem as you.
Thank you very much, now that I switched branches, the problem is solved.

meggiman, 2021-12-07:
The old pulp-sdk is no longer the recommended software environment for PULPissimo. The reference environment that we use for CI regression tests is based on the pulp-runtime (simple runtime). There is a new pulp-sdk (same repository but newer tags) that should be compatible with PULPissimo but is not automatically fetched by `make build-pulp-sdk` command. We should probably update the README and the make target at some point to align it with this newer SDK.
Other than that I cannot really tell why the core would be stuck at adress 0x00000000. It's possible that the core is stuck somewhere in the boot procedure due to old incompatible runtime initialization.

benoit1995, 2021-12-08:
> The old pulp-sdk is no longer the recommended software environment for PULPissimo. The reference environment that we use for CI regression tests is based on the pulp-runtime (simple runtime). There [...]

In fact the only thing I changed was the pulpmisso version (dde07e1).
I compiled the .bit file on this branch with the following command.
```
git checkout dde07e1
make checkout
cd fpga
make nexys rev=nexys4DDR
```
For compiling the application, I used pulp-rt-examples, and the pulp-sdk I used was the master branch version.
So I think the problem is in the .bit file, not the sdk.

rijun, 2021-12-10:
> The old pulp-sdk is no longer the recommended software environment for PULPissimo. The reference environment that we use for CI regression tests is based on the pulp-runtime (simple runtime). There [...]

Ok, thanks for the reply. I just tried out the PULPissimo master branch with the latest pulp-runtime and the runtime hello and periph examples. While it did not immediately crash, I cannot insert breakpoints (GDB error `Error accessing memory address 0x1c0080c0: Unknown error -1.`) and the core now hangs at line 137 of *crt0.S* (`137	    jal x0, pos_init_entry`).

I've compiled the application as following:
```
cd pulp-runtime
source configs/pulpissimo.sh
source configs/fpgas/pulpissimo/nexys.sh
export PATH=/opt/riscv/bin:$PATH
cd ../pulp-runtime-examples/hello
make clean all io=uart
```

Am I missing something here?

twistedpeer, 2022-01-07:
I am currently trying to port Pulpissimo to an Arty7-100T board. I have taken the FPGA implementation for the Nexys as a basis. Unfortunately I am facing exactly the same problem, i.e. I can start OpenOCD, connect GDB and also upload a program. However, unfortunately the execution does not work, with the incorrect behavior being exactly the same as described above. It did not matter if I used the Simple Runtime or the SDK.

For comparison I used the VCU108 implementation. Here, all test programs work flawlessly, regardless of whether they were built with Simple Runtime or the Pulp SDK. 

I also tried the SDK in different versions: both the version tagged 2019.12.06 as well as the current master version work fine on the VCU108, but not on the Arty7-100T. Therefore, I suspect that the problem is more in the FPGA shell of the Nexys than in the Pulpissimo RTL code.

christian-lanius, 2022-04-12:
Hi,
have you made any progress in this manner (except for using the old commit)?

Kind regards
Christian

rijun, 2022-05-14:
Hi, no, sadly not. I have since then stopped any work on this problem.