pulp-platform/pulpissimo #290: problem running `make checkout BENDER=1`

zzd1234, 2021-12-02:
Hi.

I am trying to generate source files use BENDER as indicated in README: 'Bender can be used to generate source file lists for various different tools for simulation, ASIC/FPGA synthesis etc'. 

I met following error. Would you please check that?
``` bash
$ make checkout BENDER=1
./bender update
    Fetching common_cells (git@github.com:pulp-platform/common_cells.git)
error: Dependency `common_cells` from `git@github.com:pulp-platform/common_cells.git` cannot satisfy requirement `>=1.7.4, <2.0.0`
make: *** [Makefile:54: checkout] Error 1
```
Regards

meggiman, 2021-12-07:
Hi @zzd1234,

It seems like bender is trying to fetch via ssh instead of https. Did you add your Github ssh key to your ssh-agent?
Alternatively you could try to delete the `.bender` folder and the `Bender.lock` file and try `make checkout BENDER=1` again.
Let me know if this worked  and if not what the error message is.
Best,
Manuel

zzd1234, 2021-12-09:
Thank you. This solved my problem.

Zhenda

closed 2021-12-09 by zzd1234