pulp-platform/pulpissimo #293: Error while building the RTL simulation platform

MahmouodMagdi, 2021-12-23:
I am trying to build the RTL simulation platfrom 

1. git clone https://github.com/pulp-platform/pulp
2. cd pulp
3. source setup/vsim.sh
4. make checkout 

and i get this error 

error: Failed to initialize git database

git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Makefile:41: recipe for target 'checkout' failed
make: *** [checkout] Error 1

anyone can help  me, pls?

bluewww, 2021-12-24:
This is related to github and not pulpissimo. You need to add your public key to your github account.

closed 2021-12-24 by bluewww

MahmouodMagdi, 2021-12-29:
Thanks for your help