pulp-platform/pulpissimo #294: Correct way to put C variables in shared L2 memory

JosseVanDelm, 2021-12-24:
Hi,

I'm trying to map tensor algebra workloads to a pulpissmo's HWPE, for this to work the variables have to be in shared/interleaved (non-private) L2.
Otherwise, the HWPE can not access these.
Our current workaround to this problem is to detect when variables are stored in L2 by checking their addresses in C code and copying them in memory when they are required by the HWPE. It is clear that this is a non-ideal solution.

I've been said that using linker scripts is the best way to make sure that `ld` puts the right variables in the right spots in memory.
I've found the https://github.com/pulp-platform/pulp-rt/blob/master/rules/pulpissimo/link.ld script and altered the `MEMORY_MAP` to 
```
MEMORY
{
  L2           : ORIGIN = 0x1c010004, LENGTH = 0x0006fffc
  L2_aliased   : ORIGIN = 0x00000004, LENGTH = 0x00003ffc
}
```
Basically telling the compiler to disregard the first 64 KiB of memory, as I've come to understand that this is the private bank.
This is also a workaround, not a good solution, since we now only have 448KiB available for putting stuff in memory.
I have to say I'm not entirely sure if this approach would work (I've tried it out, but I think some other bug is holding back the simulation of my C code).

So I have a few questions regarding this:
* What would be the best way to make sure the I can tell the compiler to make sure the HWPE accessed variables are accessible to the HWPE? Can this be solved by a linker script? Or should I also include some (which?) directive inside of the C code?
* Is it possible to use the linker script in the way I mentioned above?
* Inside the current linker script, what is meant with L2_Aliased? It says something about FC in the comments, but I'm not sure what FC means and what this memory region is about? Are these the addresses of other memory-mapped hardware?
* What is the best way to find out where specific variables are stored? Currently I'm just `printf`'ing their addresses in C code...
* Why is there a private region in the first place? Is this due to security or performance reasons? 

Please note that I'm not sure about a lot of stuff in my question :smile: . I find it difficult to ask the right questions, or find the right place/person to ask. Please correct me If I've stated something incorrectly above. I'm also not entirely sure what all the versions of my code/sdk/... are... This is work on top of work on top of work by other colleagues. If I need to provide more details I'll try my best to do so.
Also feel free to point me to documentation which I have overlooked, It's all a bit overwhelming to find this out by myself.

Thanks in advance! Regards!

meggiman, 2022-01-03:
Hi @JosseVanDelm 
I will try to answer a few of your questions:
1):
Yes, this can be solved by the linker script AND modifications to the C code (and it actually already is). If you have a look at the `pulp-runtime's` `data.h` file in the include directory (the legacy pulp-sdk should have a similar header definition somewhere) you will find, that it defines among many others the macro `L2_DATA`. This macro can be used to add a GCC attribute to a variable to instruct the linker to place the variable in L2 memory. E.g.

``` C
L2_DATA uint8_t hwpe_mem[256];
```

2) 
It is possible to instruct the linker to place everything in L2 memory but it should not be necessary to do so if you have a look at my answer above. 

3)
FC stands for "fabric controller". In our multi-core SoC like pulp-open which consists of a PULPissimo with attached pulp-cluster we call the RISC-V core within the PULPissimo part the "fabric controller" while the RISC-V cores within the cluster are just denoted "cluster cores". The reasoning behind this name is, that the FC within the pulp_soc.sv (thus the PULPissimo core) is usually used to program DMAs, move data from L2 memory to cluster memory (L1) or control the various on-board peripherals i.e. it *controls* the SoC *fabric* while the cluster core do the number crunching. Now why there is a memory alias for the L2 region i don't fully remember anymore. Part of it is for sure for legacy reasons. @bluewww  might know more about it.

4)
There should not be the need to figure out where a variable is stored. If you require it to be stored in a certain location, just tell the linker where you want it to be using attributes (as mentioned above). Of course, you could use pointer arithmetics (e.g. `((uint32_t) &some_var) > 0x1c0000` to check if the address of a variable is within certain bounds. But again, where the variable is placed in memory is under full control of the linker script. Therefore you should modify the linker script if at all necessary.

5)
The reason we have private memories is for performance reasons. With the current linker script in the pulp-runtime, the linker **preferably** places the text section (the  instructions) in private bank 1 and the stack data in private bank 0. If the linker placed them in L2, there will be a lot of bank conflicts (two or more master ports trying to access the same SRAM bank in the same cycle) e.g. if the next instruction and the next variable to fetch incidentally are placed on the same L2 bank. Thus, having the stack and instructions on dedicated memory banks increases the performance because there are less stalls due to the aforementioned bank conflicts. However, this does not mean, that the core really needs the instructions or stack to be placed there. If the private banks are to small for the code or the stack, the linker will just place them in the much larger L2 memory instead. However, in this case you will see the aforementioned performance penalties. 

Let me know if this helped.
Best,
Manuel

2080fresh, 2024-08-06:
> Hi @JosseVanDelm I will try to answer a few of your questions: 1): Yes, this can be solved by the linker script AND modifications to the C code (and it actually already is). If you have a look at the `pulp-runtime's` `data.h` file in the include directory (the legacy pulp-sdk should have a similar header definition somewhere) you will find, that it defines among many others the macro `L2_DATA`. This macro can be used to add a GCC attribute to a variable to instruct the linker to place the variable in L2 memory. E.g.
> 
> ```c
> L2_DATA uint8_t hwpe_mem[256];
> ```
> 
> 
> It is possible to instruct the linker to place everything in L2 memory but it should not be necessary to do so if you have a look at my answer above.
> 
> 
> FC stands for "fabric controller". In our multi-core SoC like pulp-open which consists of a PULPissimo with attached pulp-cluster we call the RISC-V core within the PULPissimo part the "fabric controller" while the RISC-V cores within the cluster are just denoted "cluster cores". The reasoning behind this name is, that the FC within the pulp_soc.sv (thus the PULPissimo core) is usually used to program DMAs, move data from L2 memory to cluster memory (L1) or control the various on-board peripherals i.e. it _controls_ the SoC _fabric_ while the cluster core do the number crunching. Now why there is a memory alias for the L2 region i don't fully remember anymore. Part of it is for sure for legacy reasons. @bluewww might know more about it.
> 
> 
> There should not be the need to figure out where a variable is stored. If you require it to be stored in a certain location, just tell the linker where you want it to be using attributes (as mentioned above). Of course, you could use pointer arithmetics (e.g. `((uint32_t) &some_var) > 0x1c0000` to check if the address of a variable is within certain bounds. But again, where the variable is placed in memory is under full control of the linker script. Therefore you should modify the linker script if at all necessary.
> 
> 
> The reason we have private memories is for performance reasons. With the current linker script in the pulp-runtime, the linker **preferably** places the text section (the instructions) in private bank 1 and the stack data in private bank 0. If the linker placed them in L2, there will be a lot of bank conflicts (two or more master ports trying to access the same SRAM bank in the same cycle) e.g. if the next instruction and the next variable to fetch incidentally are placed on the same L2 bank. Thus, having the stack and instructions on dedicated memory banks increases the performance because there are less stalls due to the aforementioned bank conflicts. However, this does not mean, that the core really needs the instructions or stack to be placed there. If the private banks are to small for the code or the stack, the linker will just place them in the much larger L2 memory instead. However, in this case you will see the aforementioned performance penalties.
> 
> Let me know if this helped. Best, Manuel

Hello, I'm working on academic research on pulpissimo. Thank you for the open-source nature.

I've found that pulpissimo has a memeory map with
- L2_PRIV0 (expected to be L1D cache 0x1c000000~0x1c008000) - 32kB
- L2_PRIV1 (L1I cache 0x1c008000~0x1c010000) - 32kB
- L2_SHARED (0x1c010000~0x1c080000)

The problem is that if I use an array or 'malloc' above a certain level, it seems the data overflows L2_PRIV0(L1D) and uses L2_PRIV1(which is the space of the PC address). So when the pulpissimo starts to read the first PC address (which is setted on ENTRY_POINT=0x1c008080), it spits an illegal instruction error on 0x1c008080.

I wonder why when the data is flushed on L2_PRIV0, it uses L2_PRIV1, not L2_SHARED. It is common that It uses L2 when the L1D is full.

I want to know the way how to use L2_PRIV0 and L2_SHARED naturally for using lots of mallocs.