pulp-platform/pulpissimo #295: Illegal Instruction Error While Running FFT code

LatifAkcayGithub, 2021-12-30:
Hello,

We are trying to run 2048 point FFT operation on Pulpissimo.
While lower number of FFT codes run successfully ( 128, 256, 1024), 2048 points version code is stuck.
We think that it may be a memory problem. So, we extend the instruction memory(PRI_1) to 128 K and Data memory (PRI_0) to 256 K.
After the memory extension, the lower FFT points again run successfully. However, 2048 points version is again stuck.
We also extend the stack size defined in link.ld from 0x800 to 0x4000. But, the code is again stuck.
In modelsim simulation, when running FFT operations, an illegal fetch adress (an unalligned adress) is generated and then the core could not find an instruction word at that adress. Then, the core starts gving "illegal instruction" error.  
So, what could be the reason behind this issue? And how can we solve it?
(We are using KissFFT C library for our fft applications)

Thank you.

bluewww, 2021-12-30:
In modelsim you can check what the address of the illegal instruction is that should tell you already a lot of what is going wrong.

LatifAkcayGithub, 2022-01-11:
Hello @bluewww 

Thanks for the answer and I am sorry for the late reply.

The issue is not related to the instruction itself. The problem is that the core generates an fetch adress which is illegal! I mean there is not any instruction at the generated adress. It seems like not alligned. Simply, the generated adress is not like 0x4 or 0x8. İt is like 0x6 ! Because of the adress is illegal, the core can not fetch any instruction from the instruction memory and then core produces "illegal instruction" error. 

When I check the problematic adress in dissassembled code, I can not see any special function or jump. İt is somewhere in a fft function.. it seems a strange error. Because when I try to make a 256 point fft, the code runs successfully. However, the error occur for 2048 points.

prajwalkashyap, 2023-06-16:
@LatifAkcayGithub Were you able to solve this issue ? If so, could you tell how did you solve it ?