pulp-platform/pulpissimo #298: Why not Benderized by default?

FrancescoConti, 2022-02-03:
Is there any specific reasons why this is not Benderized by default?

bluewww, 2022-02-03:
I guess we can enable bender by default. What do you think @meggiman ?

meggiman, 2022-02-03:
I fully agree, 
The reason it wasn't the default so far was to provide a smother transitioning period from IPApprox to Bender by keeping the existing flow the default for a while before entirely switching to Bender. Considering that the Bender flow of Pulpissimo by  now is quite mature and feature equivalent to the IPApprox flow I think we could switch with the next major release.

micprog, 2022-02-04:
I have already updated this for pulp, simply changing `ifdef BENDER` to `ifndef IPAPPROX` in the corresponding Makefiles. Let me know if you want me to open a PR for this here.
Please also note that for Bender >= v0.24, you may need to add `\"+incdir+\$$ROOT/rtl/includes\"` to the `VLOG_ARGS`, as bender's `export_include_dirs` no longer exports include directories globally, but only to the packages calling the corresponding dependency. PULP and PULPissimo have some cyclical dependency behavior regarding includes, mostly regarding `pulp_soc_defines.sv`, which is why the additional incdir above is necessary.

FrancescoConti, 2022-02-04:
@micprog 👍 I think it makes sense if you or @meggiman open a PR for this.

meggiman, 2022-02-07:
Fixed with #299

closed 2022-02-07 by meggiman