pulp-platform/pulpissimo #306: Issue in the RTL

HossamAdel11, 2022-02-20:
I have tried to collect all RTL files in one directory and I noticed that "apb_timer_unit" module is existed in 2 different directories, "\ips\timer_unit\rtl\apb_timer_unit.sv" and "\ips\pulp_soc\rtl\components\apb_timer_unit.sv", therefore I compared the 2 files and noticed that there are some differences between them !
the problem is that the "apb_timer_unit" module is used as a reference module for the "i_apb_timer_unit" instance in soc_peripherals.sv and I doesn't know whether that ref. module is the module in "\ips\timer_unit\rtl\apb_timer_unit.sv" or "\ips\timer_unit\rtl\apb_timer_unit.sv" ?

bluewww, 2022-02-21:
Thanks this is indeed a problem we  realized recently too @alex96295

bluewww, 2022-02-21:
The one in `ips/timer_unit/rtl/apb_timer_unit.sv` is the correct one.

HossamAdel11, 2022-02-21:
Okay, Thanks.

closed 2022-02-21 by HossamAdel11