pulp-platform/pulpissimo #307: FLL ip

Ahmedsaied9, 2022-02-23:
We are working on an ASIC tape-out. We have collected needed ips for the blocks we will use in the design, for the FLL block we couldn't find its ip and we figured out from prior issues that we have two options
1. We should get our own ip (which we can't afford because we are final year students working on a graduation project).
2. We can modify RTL to supply Clock externally.

For the second option does it mean that we will remove FLL ip and its signals from RTL ?, if yes, we want to know what signals we should modify.

We thought about a modification to create new input pads for SoC domain Clock & Peripheral domain Clock.

bluewww, 2022-02-23:
Yes you can basically remove the FLL instantiations and route the periph and soc clocks to pads. It is not that hard to figure out which one, just look at the outputs of the FLL instantiations.

HossamAdel11, 2022-02-24:
yes it is easy, but what about the rest of the signals that come from several modules in the PULPissimo to the FLL instances and out from them to the PULPissimo modules ? 
I think that will require a lot of work to fix this issue and keeping them as they are will be redundancy and consume power and area in the layout, not mentioned the performance that is degraded by excluding the FLLs !

bluewww, 2022-02-25:
I mean the best choice definitely still to get a hold of an FLL ip but I can't really help you with that.