pulp-platform/pulpissimo #309: Error - unexpected error from QuestaSim

ivanhira, 2022-03-03:
Hello, 

I'm trying to build the RTL simulation for QuestaSim using the command "make build" without Bender, but it throws an error when optimizing the module "riscv_cs_registers" from the file "ips/cv32e40p/./rtl/riscv_cs_registers.sv". Below are the lines containing the error. Linux (CentOS) was used in this case.
```
--Optimizing module tb_clk_gen(fast)
-- Optimizing module /home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/install/modelsim_libs/cv32e40p_lib.riscv_cs_registers(fast)
** Error: ../ips/cv32e40p/./rtl/riscv_cs_registers.sv(1228): Questa has encountered an unexpected internal error: ../../src/vlog/vgenexpr.c(1445). Please contact Questa support at http://supportnet.mentor.com/

Optimization failed
** Error: (vopt-2064) Compiler back-end code generation process terminated with code 12.
-- Optimizing package riscv_dbg_lib.dm(fast)
-- Optimizing package riscv(fast)
```
Also, I have changed the arguments in the "vopt +acc=mnpr..." command to just "vopt +acc=npr..." in the file "/home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/sim//tcl_files/rtl_vopt.tcl" (line 28 and 30) as the acc option 'm' was considered deprecated in this version of QuestaSim, which is shown below.
```
[ivan.hirata@cadence pulpissimo]$ vsim -version
Questa Sim  vsim 2021.3_2 Simulator 2021.09 Sep 10 2021
```

Full log of the "make build" command in case it is needed:
[full_log.txt](https://github.com/pulp-platform/pulpissimo/files/8181889/full_log.txt)

Thanks in advance.

bluewww, 2022-03-03:
Latest questa version crashes on some stuff in `ips/axi`. We actually fixed this in the upstream repository but have yet to import it here.

ivanhira, 2022-03-08:
Hi, sorry for the late reply.

I was trying a lot of solutions these days. Sorry if I got it wrong (I'm still a student learning about these softwares and tools), but when you said it has been fixed in the upstream repository of the ip (i believe the problem was in ``ips/cv32e40p`` instead of ``ips/axi``) I understood that it was fixed and updated in the repository of that individual IP, but the script ``./update-ips`` couldn't fetch it.
So I manually ran ``git pull`` on that folder and on ``ips/pulp_soc`` that instantiated the RISCV core. I also ran ``make scripts`` to update the scripts, but then i got the same error from the beginning in the respective file ``ips/cv32e40p/rtl/cv32e40p_cs_registers.sv``.
```
 Optimizing module /home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/install/modelsim_libs/pulp_soc_lib.soc_interconnect(fast)
-- Optimizing module /home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/install/modelsim_libs/cv32e40p_lib.cv32e40p_cs_registers(fast)
** Error: ../ips/cv32e40p/./rtl/cv32e40p_cs_registers.sv(1586): Questa has encountered an unexpected internal error: ../../src/vlog/vgenexpr.c(1445). Please contact Questa support at http://supportnet.mentor.com/

Optimization failed
** Error: (vopt-2064) Compiler back-end code generation process terminated with code 12.
-- Optimizing module /home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/install/modelsim_libs/udma_i2s_lib.i2s_rx_channel(fast)
-- Optimizing module /home/ivan.hirata/cadence_wd/616/tsmc18_miniasic17_wd/pulpissimo/install/modelsim_libs/ibex_lib.ibex_ex_block(fast)
```

bluewww, 2022-03-08:
I could reproduce it on a similar version. Looks like `-floatparameters+tb_pulp` is the culprit. You can remove this but some other stuff might break.

ivanhira, 2022-03-09:
Thank you very much! Removing ``-floatparameters+tb_pulp`` fixed the problem for now.