pulp-platform/pulpissimo #310: Error to build pulpissimo

cortalo, 2022-03-11:
Hello,

I am trying to using Bender to download the subIPs and build RTL simulations, from the clean pulpissimo.
I got the error says:
** Fatal: Read failure in vlm process (0,0)
make: *** [build] Error 141
[image]

I used to use IPApproX and everything is fine, could you help me with the problem?
Thank you!

Best regards,
Long

meggiman, 2022-03-11:
Hi @cortalo 
What version of Questasim are you using?
Also, could you show us the actual output of the commands? You piped them into less so we cannot see to full log.
This could actually be related to #309. Can you try compiling it without the `-floatparameters+tb_pulp` flag mentioned in this issue?
Best,
Manuel

cortalo, 2022-03-12:
Hi @meggiman 
Thank you for your quick reply!
I am using Queatasim-64 10.7b_1.
Actually the `Fatal: Read failure in vlm process (0,0)` is exactly due to I pipe it to less, but I quit less before it finished, thus it throw an error, sorry for that.
But still I am not able to run helloworld. Since the log is quite long, I put them in the attached files.
[make_checkout.log](https://github.com/pulp-platform/pulpissimo/files/8237236/make_checkout.log)
[make_build.log](https://github.com/pulp-platform/pulpissimo/files/8237237/make_build.log)
[helloworld_make.log](https://github.com/pulp-platform/pulpissimo/files/8237239/helloworld_make.log)

Best regards,
Long

meggiman, 2022-03-12:
Hi Long,

Try deleting your modelsim.ini file in the pulpissimo/sim directory and rerun the compilation for Questasim (make clean build). It looks like your modelsim.ini does not contain the mapping to the Questasim "work" library.
Best,
Manuel

cortalo, 2022-03-12:
Hi Manuel,

Unfortunately `make clean build` didn't solve the problem. But I found that with pulp_sdk things are working correctly on my machine. However the problem still exists when using pulp_runtime.
Thank you!
Best regards,
Long

closed 2022-03-12 by cortalo

bluewww, 2022-03-13:
It's because the bender flow requires `pulp-runtime` to symlink the `work/` directory into the `build/` folder. Currently, that patch is not in the master branch of `pulp-runtime`

bluewww, 2022-03-13:
I'll track the bug [here](https://github.com/pulp-platform/pulp-runtime/issues/29)  and get back to it when I have some time.