pulp-platform/pulpissimo #316: Error when run "make checkout"

1st-Third-Reich, 2022-03-25:
Hi all, i'm try download all the related submodules but get notice like this:

error: Failed to initialize git database in "/home/dat/pulpissimo/.bender/git/db/udma_core-41e5f4072cdae39f". Git command ("git" "fetch" "--prune" "origin") in directory "/home/dat/pulpissimo/.bender/git/db/udma_core-41e5f4072cdae39f" failed with exit code 128:

git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

make: *** [Makefile:69: checkout] Error 1

bluewww, 2022-03-25:
You need to add your public key to github

1st-Third-Reich, 2022-03-25:
Thanks, but i get new error with common_cells