pulp-platform/pulpissimo #317: Not download the related modules

1st-Third-Reich, 2022-03-25:
dat@DESKTOP-RQUNBNK:/mnt/d/pulpissimo$ make checkout
find: ‘work’: No such file or directory
curl --proto '=https' --tlsv1.2 -sSf https://fabianschuiki.github.io/bender/init \
        | bash -s -- 0.23.2
main: line 276: [: : integer expression expected
Successfully installed bender 0.23.2 in '/mnt/d/pulpissimo'.
touch bender
./bender update
     Cloning pulp_soc (https://github.com/pulp-platform/pulp_soc.git)
     Cloning tbtools (https://github.com/pulp-platform/tbtools.git)
     Cloning apb_gpio (https://github.com/pulp-platform/apb_gpio.git)
     Cloning apb (https://github.com/pulp-platform/apb.git)
     Cloning udma_external_per (https://github.com/pulp-platform/udma_external_per.git)
     Cloning adv_dbg_if (https://github.com/pulp-platform/adv_dbg_if.git)
     Cloning axi_slice (https://github.com/pulp-platform/axi_slice.git)
     Cloning timer_unit (https://github.com/pulp-platform/timer_unit.git)
     Cloning L2_tcdm_hybrid_interco (https://github.com/pulp-platform/L2_tcdm_hybrid_interco.git)
     Cloning apb_node (https://github.com/pulp-platform/apb_node.git)
     Cloning hwpe-mac-engine (https://github.com/pulp-platform/hwpe-mac-engine.git)
     Cloning ibex (https://github.com/lowRISC/ibex.git)
     Cloning common_cells (https://github.com/pulp-platform/common_cells.git)
     Cloning udma_filter (https://github.com/pulp-platform/udma_filter.git)
     Cloning tech_cells_generic (https://github.com/pulp-platform/tech_cells_generic.git)
     Cloning udma_qspi (https://github.com/pulp-platform/udma_qspi.git)
     Cloning riscv-dbg (https://github.com/pulp-platform/riscv-dbg.git)
     Cloning udma_i2c (https://github.com/pulp-platform/udma_i2c.git)
     Cloning fpnew (https://github.com/pulp-platform/fpnew.git)
     Cloning udma_core (https://github.com/pulp-platform/udma_core.git)
     Cloning jtag_pulp (https://github.com/pulp-platform/jtag_pulp.git)
     Cloning cv32e40p (https://github.com/pulp-platform/cv32e40p.git)
     Cloning apb2per (https://github.com/pulp-platform/apb2per.git)
     Cloning udma_sdio (https://github.com/pulp-platform/udma_sdio.git)
     Cloning udma_uart (https://github.com/pulp-platform/udma_uart.git)
     Cloning apb_interrupt_cntrl (https://github.com/pulp-platform/apb_interrupt_cntrl.git)
     Cloning udma_hyper (https://github.com/pulp-platform/udma_hyper.git)
     Cloning apb_adv_timer (https://github.com/pulp-platform/apb_adv_timer.git)
     Cloning axi (https://github.com/pulp-platform/axi.git)
     Cloning cluster_interconnect (https://github.com/pulp-platform/cluster_interconnect.git)
     Cloning udma_camera (https://github.com/pulp-platform/udma_camera.git)
     Cloning scm (https://github.com/pulp-platform/scm.git)
     Cloning apb_fll_if (https://github.com/pulp-platform/apb_fll_if.git)
     Cloning udma_i2s (https://github.com/pulp-platform/udma_i2s.git)
     Cloning register_interface (https://github.com/pulp-platform/register_interface.git)
     Cloning generic_FLL (https://github.com/pulp-platform/generic_FLL.git)
     Cloning common_cells (git@github.com:pulp-platform/common_cells.git)
     Cloning udma_core (git@github.com:pulp-platform/udma_core.git)
     Cloning hwpe-stream (https://github.com/pulp-platform/hwpe-stream.git)
     Cloning hwpe-ctrl (https://github.com/pulp-platform/hwpe-ctrl.git)
     Cloning tech_cells_generic (git@github.com:pulp-platform/tech_cells_generic.git)
     Cloning common_verification (https://github.com/pulp-platform/common_verification.git)
     Cloning fpu_div_sqrt_mvp (https://github.com/pulp-platform/fpu_div_sqrt_mvp.git)
touch Bender.lock
dat@DESKTOP-RQUNBNK:/mnt/d/pulpissimo$

bluewww, 2022-03-25:
I just ran `make checkout` on a freshly cloned PULPissimo. I could not reproduce your problem. Also I don't get this error you have
```
main: line 276: [: : integer expression expected
```
maybe something is up with your shell?

1st-Third-Reich, 2022-03-26:
Hi bluewww, i re-install Ubuntu but still can not download the related IPs (ips folder) and checked bash version is:

`dat@DESKTOP-RQUNBNK:~/pulpissimo$ bash --version
GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software; you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.`

closed 2022-03-27 by 1st-Third-Reich