pulp-platform/pulpissimo #320: Difficulties with custom makefile

JosseVanDelm, 2022-04-04:
Hi,

I'm trying to compile my own C code here with (version ???) of the pulp-sdk.

If I compile the C code with my own makefile on my X86 system, the binary is about 77 kB
I wish now to compile this to the pulpissimo with a version (I don't know which) of the pulp-sdk, but this is very difficult.

The makefiles that ship with my version of the pulp-sdk however are quite convoluted and difficult to comprehend.
The top makefile includes the following file:
```
include $(PULP_SDK_HOME)/install/rules/pulp.mk
```
Which subsequently calls python scripts `plpflags` and `plpconf` which create a folder with yet another makefile.
`config.mk`, `__rules.mk` and `__flags.mk`
`__flags.mk` contains a lot of flags for which I cannot find documentation.
The rules in `__rules.mk` call yet another python script: `pulp-run`.

I've tried setting `PULP_APP` and `PULP_APP_SRCS` to point to the C files and I've added the include paths to the `PULP_CFLAGS`.
This seems to work for compilation, but ultimately gives me the follwing linker error:
```
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: build/pulpissimo/demo/demo section `.bss' will not fit in region `L2'
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: address 0x1d0016a0 of build/pulpissimo/demo/demo section `.vectors' is not within region `L2'
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: address 0x1d0058cc of build/pulpissimo/demo/demo section `.l2_data' is not within region `L2'
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: address 0x1d0016a0 of build/pulpissimo/demo/demo section `.vectors' is not within region `L2'
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: address 0x1d0058cc of build/pulpissimo/demo/demo section `.l2_data' is not within region `L2'
gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/bin/ld: region `L2' overflowed by 16275660 bytes
``` 
I don't get why the linker would create a file of 16 MB, since the file on my X86 is 77Kb and the object files combined in build/pulpissimo are about 300kB (still big, but not 16 MB?)

Another solution might be to literally copy the instructions to gcc in my own makefile (from a different code example which works) but I'm afraid that this will give me problems further down the line, (when trying to run the simulator?). Is this a correct assumption?

What would be the best way to go about porting my own makefile to the pulp platform, preferably keeping all of the functionality (e.g. for simulation etc with `make run`) yet giving me finer grained control (without having to go through a ton of hoops and different scripts) of what happens with the compilation? Is there any documentation anywhere I can  consult on my version of the pulp-sdk?
I'm sorry if this sounds vague, I'd be happy to provide further detail upon request.

Any tips on how to proceed are greatly appreciated! Thanks!

bluewww, 2022-04-04:
Looks like a buggy linker script. I can't tell much more beyond that.

meggiman, 2022-04-04:
Since you mentioned that you compiled your application for x86 I assume you are not using any of the pulp drivers. In that case you might give the much simpler [pulp-runtime](https://github.com/pulp-platform/pulp-runtime) a trial. Athough the makefiles are still fairly complex the situation is still quite a bit better than with the pulp-sdk.

JosseVanDelm, 2022-04-05:
Hi @bluewww and @meggiman, thanks for the quick reply to my vague issue.

@meggiman now that you mention this it does look like I'm actually using the pulp runtime and not the pulp-sdk (my apologies).
And yes, I did not include any drivers in the x86 build, but I've been able to compile C code which includes the drivers before using the same environment (we have an absraction layer for pulp-rt in the C code that allows compilation on x86).

@bluewww  do you have any suggestions troubleshooting this linker script?
I've tried invoking the linker command from the makefile directly with `--verbose` but that did not give me much more information.
Cause I can use the same script to compile and run other examples just fine, and I did not make any modifications to `linker.ld`
The code that I'm compiling right now does include some `__attribute__((section(...)))` statements that are not in the other example, maybe there are some things I should look out for there?

Anyway, It seems like the way for me to go currently is to try to make my own makefile and proceed from there.
For either finding out the linker error or to do the actual compilation.
Thanks again for your replies!