pulp-platform/pulpissimo #325: PULP toolchain demands ETH password

LatifAkcayGithub, 2022-04-11:
Hello,

While trying to clone pulp-gcc-toolchain from https://github.com/pulp-platform/riscv-gnu-toolchain, the eth openSSH password is required. I think this is because of a confusion during the source updates. Please check it. 

Regards

bluewww, 2022-04-11:
Can you post the exact error message please? There are no git remotes pointing to ETH internal addresses.

LatifAkcayGithub, 2022-04-11:
Hello, 
please see the attached screenshot for the issue.
Thanks

[image]

bluewww, 2022-04-11:
Ah I see you copied the link from the README.md instead of cloning the repository normally. I will update the README.md

bluewww, 2022-04-11:
Fixed in https://github.com/pulp-platform/riscv-gnu-toolchain/commit/ba8d2d854895f52208d4efef7044d1c0395419d8

closed 2022-04-11 by bluewww