pulp-platform/pulpissimo #327: Packages conflicting on the AXI module version

bourquenoud, 2022-04-14:
I am unable to update nor build the system. When updating the packages, I get an conflict between two *pulp_soc* and *register_interface* as they require different versions of the *AXI* package. Selecting either one of them leads to an error.

Step to reproduce : 
- Clone the repository
- execute `make checkout`

Output :
```
touch bender
./bender update
     Cloning tbtools (https://github.com/pulp-platform/tbtools.git)
     Cloning tech_cells_generic (https://github.com/pulp-platform/tech_cells_generic.git)
     Cloning pulp_soc (https://github.com/pulp-platform/pulp_soc.git)
     Cloning common_cells (https://github.com/pulp-platform/common_cells.git)
     Cloning jtag_pulp (https://github.com/pulp-platform/jtag_pulp.git)
     Cloning common_verification (https://github.com/pulp-platform/common_verification.git)
     Cloning scm (https://github.com/pulp-platform/scm.git)
     Cloning axi_slice (https://github.com/pulp-platform/axi_slice.git)
     Cloning adv_dbg_if (https://github.com/pulp-platform/adv_dbg_if.git)
     Cloning udma_uart (https://github.com/pulp-platform/udma_uart.git)
     Cloning apb_adv_timer (https://github.com/pulp-platform/apb_adv_timer.git)
     Cloning udma_external_per (https://github.com/pulp-platform/udma_external_per.git)
     Cloning axi (https://github.com/pulp-platform/axi.git)
     Cloning apb (https://github.com/pulp-platform/apb.git)
     Cloning l2_tcdm_hybrid_interco (https://github.com/pulp-platform/L2_tcdm_hybrid_interco.git)
     Cloning cluster_interconnect (https://github.com/pulp-platform/cluster_interconnect.git)
     Cloning udma_camera (https://github.com/pulp-platform/udma_camera.git)
     Cloning udma_sdio (https://github.com/pulp-platform/udma_sdio.git)
     Cloning apb2per (https://github.com/pulp-platform/apb2per.git)
     Cloning udma_core (https://github.com/pulp-platform/udma_core.git)
     Cloning udma_hyper (https://github.com/pulp-platform/udma_hyper.git)
     Cloning timer_unit (https://github.com/pulp-platform/timer_unit.git)
     Cloning udma_qspi (https://github.com/pulp-platform/udma_qspi.git)
     Cloning cv32e40p (https://github.com/pulp-platform/cv32e40p.git)
     Cloning apb_fll_if (https://github.com/pulp-platform/apb_fll_if.git)
     Cloning udma_i2s (https://github.com/pulp-platform/udma_i2s.git)
     Cloning apb_node (https://github.com/pulp-platform/apb_node.git)
     Cloning apb_gpio (https://github.com/pulp-platform/apb_gpio.git)
     Cloning generic_fll (https://github.com/pulp-platform/generic_FLL.git)
     Cloning udma_filter (https://github.com/pulp-platform/udma_filter.git)
     Cloning riscv-dbg (https://github.com/pulp-platform/riscv-dbg.git)
     Cloning register_interface (https://github.com/pulp-platform/register_interface.git)
     Cloning fpnew (https://github.com/pulp-platform/fpnew.git)
     Cloning apb_interrupt_cntrl (https://github.com/pulp-platform/apb_interrupt_cntrl.git)
     Cloning udma_i2c (https://github.com/pulp-platform/udma_i2c.git)
     Cloning hwpe-mac-engine (https://github.com/pulp-platform/hwpe-mac-engine.git)
     Cloning ibex (https://github.com/pulp-platform/ibex.git)
     Cloning hwpe-ctrl (https://github.com/pulp-platform/hwpe-ctrl.git)
     Cloning hwpe-stream (https://github.com/pulp-platform/hwpe-stream.git)
     Cloning fpu_div_sqrt_mvp (https://github.com/pulp-platform/fpu_div_sqrt_mvp.git)
Requirement `>=0.35.1, <0.36.0` conflicts with other requirements on dependency `axi`.

- package `pulp_soc` requires `>=0.29.1, <0.30.0`
- package `register_interface` requires `>=0.35.1, <0.36.0`

To resolve this conflict manually, select a revision for `axi` among:
0) `>=0.29.1, <0.30.0`
1) `>=0.35.1, <0.36.0`
Enter a number or hit enter to abort: 
```

bluewww, 2022-04-14:
Fixed #326

closed 2022-04-14 by bluewww

bourquenoud, 2022-04-14:
That was unexpectedly fast. Thanks, the error doesn't show up anymore.