pulp-platform/pulpissimo #330: Make checkout fail

gnss-tao, 2022-04-20:
Hi, when I tried to use make check to fetch all the required source. I had the following error message.  I am using the latest master version (ba01e616c15352c43eb73e28400942ef66ade386).

```
make checkout
find: ‘work’: No such file or directory
./bender checkout
    Checkout cluster_interconnect (https://github.com/pulp-platform/cluster_interconnect.git)
    Checkout cv32e40p (https://github.com/pulp-platform/cv32e40p.git)
    Checkout ibex (https://github.com/pulp-platform/ibex.git)
    Checkout l2_tcdm_hybrid_interco (https://github.com/pulp-platform/L2_tcdm_hybrid_interco.git)
    Checkout register_interface (https://github.com/pulp-platform/register_interface.git)
    Checkout riscv-dbg (https://github.com/pulp-platform/riscv-dbg.git)
    Checkout scm (https://github.com/pulp-platform/scm.git)
    Checkout timer_unit (https://github.com/pulp-platform/timer_unit.git)
    Checkout udma_camera (https://github.com/pulp-platform/udma_camera.git)
    Checkout udma_external_per (https://github.com/pulp-platform/udma_external_per.git)
    Checkout udma_filter (https://github.com/pulp-platform/udma_filter.git)
    Checkout udma_hyper (https://github.com/pulp-platform/udma_hyper.git)
    Checkout udma_i2c (https://github.com/pulp-platform/udma_i2c.git)
    Checkout udma_i2s (https://github.com/pulp-platform/udma_i2s.git)
    Fetching udma_i2s (https://github.com/pulp-platform/udma_i2s.git)
    Checkout udma_qspi (https://github.com/pulp-platform/udma_qspi.git)
    Fetching udma_qspi (https://github.com/pulp-platform/udma_qspi.git)
    Checkout udma_sdio (https://github.com/pulp-platform/udma_sdio.git)
    Fetching udma_sdio (https://github.com/pulp-platform/udma_sdio.git)
    Checkout udma_uart (https://github.com/pulp-platform/udma_uart.git)
    Fetching udma_uart (https://github.com/pulp-platform/udma_uart.git)
    Checkout pulp_soc (https://github.com/pulp-platform/pulp_soc.git)
    Fetching pulp_soc (https://github.com/pulp-platform/pulp_soc.git)
    Checkout tbtools (https://github.com/pulp-platform/tbtools.git)
    Fetching tbtools (https://github.com/pulp-platform/tbtools.git)
error: Git command ("git" "tag" "bender-tmp-7d0a4f8acae71a583a6713cab5554e60b9bb8d27" "7d0a4f8acae71a583a6713cab5554e60b9bb8d27" "--force") in directory "/mnt/pulpissimo/.bender/git/db/cluster_interconnect-b387c598c6b7108b" failed with exit code 128:

fatal: cannot update ref 'refs/tags/bender-tmp-7d0a4f8acae71a583a6713cab5554e60b9bb8d27': trying to write ref 'refs/tags/bender-tmp-7d0a4f8acae71a583a6713cab5554e60b9bb8d27' with nonexistent object 7d0a4f8acae71a583a6713cab5554e60b9bb8d27

make: *** [Makefile:69: checkout] Error 1
```
Thanks

meggiman, 2022-04-20:
Hi @gnss-tao ,

Can you try deleting the `Bender.lock` and the hidden `.bender` directory? Also: Are you using the latest version of PULPissimo? I cannot reproduce your error with a fresh clone of PULPissimo master.
Best,
Manuel

BEforlin, 2022-05-24:
Hi all,

I have reproduced this error by using a fresh Ubuntu install where curl is not installed. 

It will complain that curl is not installed, and won't work afterward. Deleting `Bender.lock` and `.bender` did not solve the problem. 

I fixed it by deleting and cloning the rep again, it will work now that curl is installed.

Best,
Bruno

chengquan, 2022-08-12:
Hi BEforlin,
I meet the same problem too. 
I try to solve it according to your answer.
But it still does not work.
So,  is there any other solution? I use three different PCs with different networks to test it. They all meet the same problem.
Best,
Quan

chengquan, 2022-08-12:
Hi all,
Just making a new dir named "work" in "sim" folder can solve this problem.....
So easy....lol....

bluewww, 2022-08-12:
Thanks for the feedback. Looks like there might be a mismatch betwen used pulp-runtime version and pulpissimo