pulp-platform/pulpissimo #331: Running a helloworld on Genesys2

RuokaiYin, 2022-04-20:
Hi there,

We are trying to run a simple helloworld program on the genesys2 board. Since now, we managed to compile and write the bitstream file of pulpissimo with the RI5CY core on to the genesys board. Now we are stucking at compiling and writing the helloworld program to the board.
Could you give us some help on what detailed steps should we begin with? We did check the instruction inside the README file, but were confused with some steps. For example, after we create a hello.c file with specified board frequencies, where should we put the file and compile it? Do we need anything extra from the pulp-SDK to compile the codes? When we write the program to the board through the USB JTAG adapter, which pin on the genesys2 corresponding to the JTAG port of the pulpissimo?

Sorry for a lots of questions! Looking forward to hear from you.

Best,
Ruokai

meggiman, 2022-05-02:
1. Use the latest version of pulp-runtime (the legacy PULP SDK is no longer supported). You can find it [here](https://github.com/pulp-platform/pulp-runtime). 
2. After cloning it (I assume you already installed the RISC-V GCC toolchain) you cd into `pulp-runtime` and type `source configs/fpga/pulpissimo/gensys2.sh`. Now your terminal is ready for compilation of C code for PULP (if you close the terminal or switch to a different terminal window, you need to repeat step 2).
3. In the same terminal, switch to your application directory (I suggest you start with the hello world example in the [pulp-runtime-examples](https://github.com/pulp-platform/pulp-runtime-examples).
4. Now start the compilation with `make clean all io=UART`. This will build compile the application and link the ELF binary. (The io=uart tells the runtime to redirect stdout (printf output) to the UART peripheral). By default the binary is written to `build/test/test`
5. In a new terminal, start OpenOCD with the gensys2 config file (can be found [here](https://github.com/pulp-platform/pulpissimo/blob/master/fpga/pulpissimo-genesys2/openocd-genesys2.cfg)). i.e. type:
```
openocf -f fpga/pulpissimo-genesys2/openocd-genesys2.cfg
```
6. In a new terminal, start RISC-V GDB pointing to the generated ELF binary:
```
riscv32-unknown-elf-gdb  <C code dir>/build/test/test
```
7. In the GDB shell, connect to your OpenOCD server (which runs in the other terminal):
```
target remote localhost:3333
```
8. Load the binary into PULPissimos main memory and run it
```
load
continue
```

Most of these steps are described pretty clearly in the README ([here](https://github.com/pulp-platform/pulpissimo#gdb-and-openocd)). The only thing that is different (we should update the README in that regard) is that you should use `pulp-runtime` rather than `pulp-sdk`.

Best,
Manuel

RuokaiYin, 2022-05-02:
Thank you for the response! This is very helpful information, and I end up managing to run the codes on the board! Many thanks again!

closed 2022-05-02 by RuokaiYin

Wen-Sheng, 2023-04-17:
HI @meggiman, 
I followed your flow but ran into the following problem. . . I don’t know how you solved it, my environment is ububtu18.04. . .

-------------------------------------------------------------------------------------------------------------
ubuntu@ubuntu:~/pulp/pulpissimo/pulp-runtime-examples/hello$ make clean all io=UART
RM  /home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build
rm -rf /home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build
CC  test.c
mkdir -p `dirname /home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build/test/test.o`
/home/ubuntu/riscv_compiler/pulp-riscv32/bin/riscv32-unknown-elf-gcc -c test.c -o /home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build/test/test.o -MMD -MP -march=rv32imcxgap9 -O3 -g -D__riscv__ -fdata-sections -ffunction-sections -include chips/pulpissimo/config.h -I/home/ubuntu/pulp/pulpissimo/pulp-runtime/include/chips/pulpissimo -DARCHI_FPGA_PER_FREQUENCY=10000000 -DARCHI_FPGA_SOC_FREQUENCY=20000000 -fno-jump-tables -fno-tree-loop-distribute-patterns -D__PLATFORM__=ARCHI_PLATFORM_FPGA -DCONFIG_IO_UART=0 -DCONFIG_IO_UART_BAUDRATE=115200 -DCONFIG_IO_UART_ITF=0 -I/home/ubuntu/pulp/pulpissimo/pulp-runtime/lib/libc/minimal/include -I/home/ubuntu/pulp/pulpissimo/pulp-runtime/include -I/home/ubuntu/pulp/pulpissimo/pulp-runtime/kernel
internal error: can't hash `pv.add.h.div2': exists
Assembler messages:
Fatal error: Broken assembler.  No assembly attempted.
/home/ubuntu/pulp/pulpissimo/pulp-runtime/rules/pulpos/default_rules.mk:260: recipe for target '/home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build/test/test.o' failed
make: *** [/home/ubuntu/pulp/pulpissimo/pulp-runtime-examples/hello/build/test/test.o] Error 1

Wen-Sheng, 2024-07-10:
This project was implemented last year, so it was a long time ago...
I remember, I ran the command `make clean all run`
If there is output on UART, I think I modified the code myself...

lxxzxxjxx, 2024-07-22:
> This project was implemented last year, so it was a long time ago... I remember, I ran the command `make clean all run` If there is output on UART, I think I modified the code myself...

Thanks for you reply，I solved it by changing the gcc compiler with pulp itself