pulp-platform/pulpissimo #333: make checkout failed

neekoli, 2022-04-26:
The image below shows the v7 pulpissimo after "make checkout", with "ips" "iptools" added. 
[image]
However, when I download the latest version of "make checkout", I tried  "make checkout" to get other ips. But 2 problems occurred as follows.

- axi version conflict
- no ips were cloned successfully

[image]
[image]
(The image below shows the latest verison pulpissimo after "make checkout", with no "ips" "iptools" added. )
[image]

bluewww, 2022-04-26:
The upgrade path from older version is unfortunately not well supported due to the tooling changes we did lately (deprecating and removing ipstools).

When I run `make checkout` on a freshly cloned pulpissimo I don't get any errors.

neekoli, 2022-04-26:
> The upgrade path from older version is unfortunately not well supported due to the tooling changes we did lately (deprecating and removing ipstools).
> 
> When I run `make checkout` on a freshly cloned pulpissimo I don't get any errors.

Thank you. But I did run "make checkout" on a freshly cloned pulpissimo today in Ubuntu 18.04.

micprog, 2022-04-26:
For the `axi` version issue, you can also just select one of the `axi` versions. I know the `v0.29.1` version is tested, but there shouldn't be any major conflicts with the newer versions. The `axi` dependency will be updated in upcoming versions to avoid manual selection when re-resolving dependency versions. I'm unsure why bender is re-resolving dependencies on your system when running `make checkout`, if the `Bender.lock` file is available this should not happen.

For the IPs, with the new dependency management, no `ips` or `ipstools` folder will be created, as dependencies are now managed with bender. Please check the readme https://github.com/pulp-platform/pulpissimo#where-are-all-the-sub-ips-dependencies for more details.

neekoli, 2022-04-27:
> For the `axi` version issue, you can also just select one of the `axi` versions. I know the `v0.29.1` version is tested, but there shouldn't be any major conflicts with the newer versions. The `axi` [...]

Oh I see！I truly appreciate your timely help~

closed 2022-04-27 by neekoli