pulp-platform/pulpissimo #343: can't write L2 RAM via JTAG (nexys4 FPGA)

rem3-1415926, 2022-05-26:
I have set up a VM with the simple runtime and the pulpissimo repo (both the current state and release 7.0.0) and I can seemingly correctly build and download the SoC on the nexys4 board (not video, DDR or A7. That mistake lays already in the past). 
However, I can't write anything into 0x1c00 0000 and above, where the L2 RAM should be according to the documentation and the linker scripts. So far I've tried:
- reading pc register: seems to work
- setting pc register: seems to work
- running the program (continue) in ROM: may or may not work, but pc remains constant as expected
- reading memory mapped periphery: seems to work
- writing memory mapped periphery: seems to work (can read back what I've written, as long as the register supports it)
- reading above memory space: yields "badcable" (badcab1e)
- writing to L2 memory manually: no direct feedback, as usual
- writing program to L2 memory ("load" command): prints out the section it loads, at 33KB/sec, 268bytes/write, no obvious sign of failure
- reading L2 memory: yields 0x0 everywhere, including where I wrote to immediately before
- disass main: also reads 0x0 everywhere (displayed as "unimp" instruction)
- naturally, I still tried to run it - it crashes, of course (pc reads 0x0 after ctrl+C termination)

So I conclude that it's just the RAM that cannot be written to for whichever reason. I have noticed that at some point you have removed the out of context IPs for BlockRAM, is it possible that something went wrong there, or am I missing something on my end? 

I pretty much followed the instructions in the readmes of pulpissimo and pulp-runtime straight forward, minus the RTL simulation platform that I skipped and the FPU that I disabled. Both the UART send and the hello world example appear to have compiled to something useful (definitely not the series of 0x0 I'm reading back from the FPGA)

christian-lanius, 2022-06-29:
I think the broken Nexys 4 project is a known issue, c.f. #289

rem3-1415926, 2022-06-29:
I thought I looked thoroughly for already existing issues on this, but apparently I missed #289. 

However, I've actually gotten it to run by now - though I haven't figured out why exactly. I set up the entire thing anew, on a Redhat server with Vivado 2022.2 (I believe?), instead of the Ubuntu VM on my notebook and Vivado 2020.2 that I had before. Doing so revealed that the standard RAM configuration didn't quite fit into the available blockram (no warning or error on the previous setup), and after reducing the number of interleaved memory banks from 4 to 2, the above described issues seem to be solved. 

Not sure if *everything* runs as it's supposed to, I haven't gotten the timer interrupts to work yet (timer itself seems to run). But then again, I haven't spent much time on that so I'm currently suspecting a user error on my end. That's the state as of right now. Not as conclusive as I'd like it to be, but since it will take some time until I get back to it, I figured I'd put up this quick update anyway.