pulp-platform/pulpissimo #351: Illegal instruction "hello" example

mathijskr, 2022-06-14:
Hello,

I am trying to run the hello example from the pulpissimo-runtime-examples repository in vsim. Compilation of both the pulpissimo soc and the hello example was successful, but running the simulation in vsim produces illegal instructions. I followed all instructions from the pulpissimo repo and I am using the prebuilt toolchain from here: https://github.com/pulp-platform/riscv-gnu-toolchain/releases/tag/v2.5.0.

The simulations starts with several driver issues (such as this one: # ** Warning: (vsim-8683) Uninitialized out port /tb_pulp/i_dut/soc_domain_i/pulp_soc_i/i_clk_rst_gen/i_fll_cluster/i_FLL_digital/JTQ has no driver.)

Then it gives this output:
# [PULP] 1110801ns - WRITE32 burst @1c000000 for        1024 bytes.
# [PULP] 1940901ns - WRITE32 burst @1c000400 for        1024 bytes.
# [PULP] 2227001ns - WRITE32 burst @1c000800 for         344 bytes.
# [PULP] 3057101ns - WRITE32 burst @1c008000 for        1024 bytes.
# [PULP] 3189601ns - WRITE32 burst @1c008400 for         152 bytes.
# [TB  ] 3191201ns - Resuming the CORE
# [TB  ] 3235301ns - retrying debug reg access

And then:
# 4062654ns: Illegal instruction (core 0) at PC 0x1c008476:
# 4062752ns: Illegal instruction (core 0) at PC 0x00000000:
# 4062752ns: Illegal instruction (core 0) at PC 0x00000000:
...

Could someone please point me in the right direction to solve this issue?
Thanks,
Best regards,
Mathijs

Marcello-Barbirotta, 2022-07-19:
Hello,
Same issue for me. Looking the dump file, the first illegal instruction is a csrw in "pos_irq_init" routine.

[image]

Any news from that?

Thanks

bluewww, 2022-07-19:
This is `#define CSR_STACK_START  0x7D1`, a csr that doesn't exist in PULPissimo. What were the commands you used to compile the hello example? Which configuration script did you source?

Marcello-Barbirotta, 2022-07-19:
Hi blueww.
Following the guide, i source the "pulpissimo/pulp-runtime/configs/pulpissimo_ibex.sh" and i compile with "make clean all run"

bluewww, 2022-07-19:
Are you using the ibex core? If not you need to source the cv32 related script.

Marcello-Barbirotta, 2022-07-19:
With an old toolchain, both ibex and cv32 gives the same error.
Using the updated toolchain, cv32 works fine while ibex was giving the same error before.

bluewww, 2022-07-19:
the old toolchains doens't support the cv32 so that is expected.
the issue concering ibex is a genuine bug. @micprog  do you have this csr in ibex?

dasbacete, 2022-12-11:
I have the same problem. Is this solved? I need to work with the ibex and Pulpissimo.

prajwalkashyap, 2023-06-16:
@Marcello-Barbirotta  What is the updated toolchain you are referrring to ?

Marcello-Barbirotta, 2024-01-18:
Hi @prajwalkashyap.
Coming back to that problem, recently i installed again the pulpissimo environment trying two different toolchain in two different repositories you have (the second one should be the latest) :
- https://github.com/pulp-platform/pulp-riscv-gnu-toolchain
- https://github.com/pulp-platform/riscv-gnu-toolchain
Both cases give me the same problem.
I can run hello world with cv32 but the same error comes up when I try with ibex.

Tommao12345, 2025-06-19:
I have the same problem. Is this solved?