pulp-platform/pulpissimo #353: Interrupt from HWPE

christian-lanius, 2022-06-29:
I want to add an accelerator in the SoC, so I enable the HWPE with the respective flag in the testbench and I can successfully run the example accelerator as well as the datamover example (with some minor changes), however, the wait for the accelerator does not work as far as I can tell:
In the example HWMAC code (https://github.com/pulp-platform/hwpe-tb/blob/master/sw/tb_hwpe.c#L73) a wfi instruction is used to pause the program until the accelerator is done. However, in my case the code never continues beyond that point. I see the insn in the tracer appear before the event triggers, but then nothing happens.
[image]
This happens with the example code and the MAC engine, as well as with a minimal example with the datamover example.
My understanding from the RISC-V spec is that I don't have to enable any interrupts or anything for this to work: "The Wait for Interrupt instruction (WFI) provides a hint to the implementation that the current hart can be stalled until an interrupt might need servicing [...] The WFI instruction can also be executed when interrupts are disabled."
Even forcing the irq_i input of the core to a non zero value, the core does not continue execution and the interrupt is not acknowledged by the core.
Additionally, just looking at the code, it appears as if the HWPE interrupt reside in the s_events vector in soc_peripherals at 140+141 [code](https://github.com/pulp-platform/pulp_soc/blob/master/rtl/pulp_soc/soc_peripherals.sv#217). However the soc_event_generator is parameterized to expect them at location 7+8 [code](https://github.com/pulp-platform/pulp_soc/blob/master/rtl/pulp_soc/soc_peripherals.sv#L628) as far as I can tell, or is that 

Is there an example piece of code or some kind of test I can perform to see how interrupts are working?

christian-lanius, 2022-06-29:
I have found a way to wait for the interrupt, specifically what needs to be done is:

- Enable the interrupt in the c-code (`itc_mask_set_set(ARCHI_FC_ITC_ADDR, 0x40000000);`)
- There is a difference between the documentation [p.9](https://github.com/pulp-platform/pulpissimo/blob/master/doc/datasheet/datasheet.pdf) and the pulp runtime [link](https://github.com/pulp-platform/pulp-runtime/blob/master/include/archi/chips/pulpissimo/memory_map.h#L36), where to interrupt controller is actually located in the memory map. The value from the documentation seems to be correct (0x00009000), so I changed that in the memory_map.h file
- Change in the verilog code the parameter pointing to the index of the hwpe in the event vector from 7 to 140 [link](https://github.com/pulp-platform/pulp_soc/blob/master/rtl/pulp_soc/soc_peripherals.sv#L628)
- Somehow force the mie_bypass signal to be set for this interrupt. I tried to do this in c, but I am currently unable to do so. As far as I can tell, it is controlled by the CSR registers, specifically 0x304. Forcing this register in simulation to the correct value, everything works.

[image]

Regarding the last point, I tried the naive way of doing this in c
```
volatile uint32_t *mie_reg;
mie_reg = 0x304;
*mie_reg = 0x40000000;
```
but that did not work for some reason. Is the offset for these registers wrong and if so, where can I find the full map with the correct offsets?

bluewww, 2022-06-29:
The FC_ITC (apb_interrupt_cntrl) is visible from two addresses due to aliasing that is right. I think we often used 0x00009800 but looking at the rtl 0x00009000 makes more sense. Both should work.