pulp-platform/pulpissimo #354: Synthesis error on soc_interconnect.sv:277 (board: nexys4DDR and nexys_Video)

yadi14, 2022-07-15:
I am new to pulpissimo and try to run the bitstream generation with "make <board_target>" command. I tried both nexys4DDR and nexys_Video options but facing the same error as shown below. I also tried on two version of Vivado, 2019.2 and 2022.1, both give the same message.

`ERROR: [Synth 8-9123] an enum variable may only be assigned the same enum typed variable or one of its values [.../pulpissimo/.bender/git/checkouts/pulp_soc-125142425fefd4e5/rtl/pulp_soc/soc_interconnect.sv:277]`

Is there any workaround way to this enum variable issue?
Greatly appreciate your help!

tgouss, 2022-07-26:
Hello @yadi14, 
I also have to tackle this issue. I solve it by insert before line 388 in ./ips/axi/axi/src/axi_pkg.sv
`MY_CUT        = MuxAw | MuxAr | MuxW,`

 After that, I rewrite the line 277 in ./ips/pulp_soc/rtl/pulp_soc/soc_interconnect.sv
`LatencyMode: axi_pkg::MY_CUT, `

It looks like Vivado wasn't able to understand that it is a correct assignment by the previous way.
I wonder if there is not a solution by keeping `axi_pkg::CUT_MST_AX | axi_pkg::MuxW` and just editing the rest of the line 279.

@bluewww maybe there is more elegant way to solve it?

Hope this will help you,

yadi14, 2022-08-12:
Thanks, @tgouss ! It works!
Greatly appreciate your help!

closed 2022-08-12 by yadi14

tgouss, 2022-08-12:
Your welcome!
This fix could be implemented cleanly on a next PR.

davidp135, 2023-02-22:
Hi, it looks like this issue still exists when building the nexys video in Vivado 2022.2. Is there a plan to merge the changes above into pulpissimo? I am happy to help if the changes above are acceptable.

haresh-seenivasagan, 2023-11-17:
hi @tgouss  @yadi14 on what file are these changes supposed to be made ? . i cant find them on the pulpissimo repo

tgouss, 2023-11-17:
Hi @haresh-seenivasagan,

You should first execute `./update-ips` which clone axi IP.

If you get `No module named 'ipstools' error, please first build ipstools by running `make ipstools`.
``` bash
from ipstools import ipstools
ModuleNotFoundError: No module named 'ipstools'
```

Note: I used v7.0.0 release.

I hope it helps you.