pulp-platform/pulpissimo #357: Bypass FLL (with pulp-freertos)

ivanhira, 2022-10-14:
Hi,

I'm trying to use external clock signals for the SoC and Peripheral domains. I've done this before when PULPissimo was still using pulp-sdk, removing the FLL modules and routing the clocks from the pads created to the module "soc_clk_rst_gen.sv", and in the sdk in "init.c" file, I commented the __rt_freq_init() function and set the clocks manually with the function __rt_freq_set_value().
I tried the same method in freeRTOS but it's not working, the simulation stops (it doesn't break) at "Waiting for end of computation" and runs forever, so i guess it's waiting for some signal from the removed FLLs.
Is there any other way or setup to implement this?

Thanks in advance.

bluewww, 2022-10-14:
You can check this configuration file how to disable clock configuration:
```
https://github.com/pulp-platform/pulp-freertos/blob/541fae79cfbd7c02d55024887420b315b8f6d73c/env/fpga.sh
```
Basically you need to these environment variables when compiling:
```
export CONFIG_DRIVER_FLL=n
export CONFIG_DRIVER_CLKDIV=n
export CONFIG_DRIVER_CLKCONST=y
```

ivanhira, 2022-10-14:
Thanks for the fast reply. 

I had to change some lines in some source files, but it worked. Thanks.

closed 2022-10-14 by ivanhira