pulp-platform/pulpissimo #359: Questasim not picking up the include files in common_cells-cfda49247f806ee2/include

sammy17, 2022-11-04:
This may be a Questasim issue rather than a pulpissimo issue, but I wanted to check whether any of you have encountered this and/or have a workaround. 

I am getting an error while compiling pulpissimo using Questa Sim 2020.2_2:
common_cells-cfda49247f806ee2/src/isochronous_4phase_handshake.sv(41): Cannot find `include file "common_cells/registers.svh" in directories:

I can see that **common_cells-cfda49247f806ee2/include** is included as an include dir and tool looks at this path but somehow it cannot find the registers.svh file in **common_cells/registers.svh** (I can see the file there). 

Please let me know if anyone encountered this error before and has a workaround.

meggiman, 2022-11-04:
Hi @sammy17 

Unfortunately, I don't have questa 2020.2 installed. But neither with version 2020.1 nor with version 2021.1 I can replicate your issue on a fresh clone of pulpissimo `master`. How did you generate the scripts for compilation? Did you just run the `all` make target? Did you inspect the `sim/compile.tcl` that Bender generates to see if it supplies the correct incdir arguments when compiling the common cells?

sammy17, 2022-11-04:
Hi @meggiman 

Yeah, I just ran `make build` target in the root directory which invoked `make all` in sim directory, hope I am not missing any steps. 
I checked the compile.tcl, it looks correct. 

```
if {[catch {vlog -incr -sv \
    -suppress 2583 -suppress 13314 "+incdir+$ROOT/rtl/includes" \
    +define+TARGET_RTL \
    +define+TARGET_SIMULATION \
    +define+TARGET_TEST \
    +define+TARGET_VSIM \
    "+incdir+$ROOT/.bender/git/checkouts/common_cells-cfda49247f806ee2/include" \

```
It includes the correct directory, I tried including the `$ROOT/.bender/git/checkouts/common_cells-cfda49247f806ee2/include/common_cells` directory as well after changing the ``include` statement to ``include "registers.svh"`, but that did not work as well. 

It should be a tool issue as you are not getting the error, I will check if I have access to any other versions of the tool. 
If not, will try to debug this further. 
Thanks for the quick response, please let me know if I am missing any steps.

sammy17, 2022-11-05:
The error was not there with an older Model Sim version, so it is confirmed it's a (weird) tool issue. Therefore, I will close this issue. Thanks.

closed 2022-11-05 by sammy17