pulp-platform/pulpissimo #360: rt/rt_api.h File not found

sammy17, 2022-11-07:
I am trying to run tests with pulpissimo SoC ibex version and I am getting the following error when I try to compile the GPIO input test. 

`test.c:7:10: fatal error: rt/rt_api.h: No such file or directory`

I see that many tests include this file but I could not find this file in pulp-runtime repo or pulpissimo repo. Where can I find this file? 
I am using the pulpissimo release v7.0.0.

meggiman, 2022-11-07:
Hi @sammy17 
What GPIO input test are you relating to?

sammy17, 2022-11-07:
Hi @meggiman 
I am referring to pulp-rt-examples/gpio/input/test.c testcase in the repo https://github.com/pulp-platform/pulp-rt-examples

meggiman, 2022-11-07:
These examples are a remnant of the old legacy pulp-sdk and they are not compatible with pulp-runtime nor the new pulp-sdk. Do you specifically need an example of how to use the GPIOs or is it just a matter of trying some sample application on the platform?

meggiman, 2022-11-07:
In that case have a loot at our regression test repository. It contains a number of different tests for the peripherals that can also act as an example on how to use them. You can find the repo here:
https://github.com/pulp-platform/regression_tests

sammy17, 2022-11-07:
Thanks, that was really helpful! 
One last question, do you have tests for the HWPE accelerator? 
I did not see tests for it in the regression tests.

meggiman, 2022-11-09:
I don't. I am not sure there are any to be honest. The hwpe module in PULPissimo is just intended as an example module for the HWPE infrastructure and not intended to be used as a real accelerator. @FrancescoConti do you know if there are any?