pulp-platform/pulpissimo #379: Compile pulpissimo on zcu104 board:Timing constraints are not met

alruppin, 2022-12-21:
Hi
I attempted to compile Pulpissimo on ZCU104. It ends up with timing constraints aren't met:
Current Timing Summary | WNS=-7.567 | TNS=-29782.275 
The procedure I followed is:
1.	download pulpissimo-master.zip from https://github.com/pulp-platform/pulpissimo
2.	cd <my_path>/new_pulpissimo/
3.	Copy pulpissimo-master.zip into here
4.	unzip pulpissimo-master.zip
5.	cd <my_path>/pulpissimo-master
6.	make checkout
7.	make scripts
8.	cd <my_path>/pulpissimo-master/fpga
9.	make zcu104
OS is Ubuntu 18.04, Vivado 2020.2
Can you please assist? What I'm doing wrong?
Thx

christian-lanius, 2022-12-22:
Normally, such large WNS correspond to paths which are not real timing paths, but which have no timing exceptions defined.
Can you share the timing report in text form here?

alruppin, 2023-01-04:
Added pulpissimo-zcu104.timing_WORST_100.rpt
Best regards,

meggiman, 2023-01-10:
Hi @alruppin,
Where did you add this report? Without this information, it is almost impossible to help.

alruppin, 2023-01-10:
[pulpissimo-zcu104.timing_WORST_100.rpt.gz](https://github.com/pulp-platform/pulpissimo/files/10382872/pulpissimo-zcu104.timing_WORST_100.rpt.gz)
Hi 
Please let me know if see upload? also I'll send it be email.
Best regards,

alruppin, 2023-01-10:
Hi
Attached to this email.
I’ll try upload it thru github too.

Best regards
Alex

alruppin, 2023-01-12:
[pulpissimo-zcu104.timing_WORST_100.txt](https://github.com/pulp-platform/pulpissimo/files/10397141/pulpissimo-zcu104.timing_WORST_100.txt)
HI, I've uploaded the report for the third time. Please confirm that you have it.
Best regards,
Alex

alruppin, 2023-01-13:
Hi
Only after modifying to 100 NS, the timing is O.K
export FC_CLK_PERIOD_NS=50                                    | export FC_CLK_PERIOD_NS=100.
Here I have couple of questions:
What is your release process? Shouldn't such stuff be caught?
I believe that issue is caused in the first place, because of pulp_clock_gating_xilinx.sv. Which you use instead of BUFGCE, it could be justified for small FPGA, but not ZCU104. It has of BUFGCE. I did an experiment replacing all this cells, but one with BUFGCE and all is good without reducing frequency.
Will simulation run under VIVADO simulator?(after adapting of all scripts)?
What kind of JTAG probe you recommend? I see examples of: digilent-jtag-hs1.cfg and olimex-arm-usb-ocd-h.cfg. Will it work with digilent JTAG HS2 or digilent JTAG HS3?
Thank you,
Alex

bluewww, 2023-01-13:
The problem is that we have several FPGA ports but not enough manpower to maintain them all to the same standard as the genesys2 (which is the one we use mainly). We rely on contributions such as yours to improve FPGA ports. In fact all FPGA ports except the genesys2 one are user-contributed.

We would be happy to take any patches that improves and fixes issues on the zcu104. Could you turn your changes into a pull request please?

alruppin, 2023-01-15:
I'll be more than happy to share patches with you, but first I want to try it on ZCU104. 
I'll update you when results are available.
Could you please  answer my question? 
What kind of JTAG probe you recommend? 
I see examples of: digilent-jtag-hs1.cfg and olimex-arm-usb-ocd-h.cfg. 
Will it work with Digilent JTAG HS2 or Digilent JTAG HS3?
Best regards,
Alex

bluewww, 2023-01-16:
I have not tested other JTAG probes but I don't see why the HS2 or HS3 couldn't be made to work (obviously needs a tweaked *.cfg file).

Maybe @meggiman knows more.

meggiman, 2023-01-16:
Hi @alruppin,
I know for a fact that the digilent hs2 cable works. Also, looking at the configuration scripts available in the openOCD source tree it seems like the HS3 is also supported (at least there is a configuration file for it). However you will have to take a look at the existing OpenOCD configuration file in the fpga folder and adapt the first couple of lines if you use a different jtag cable. You will find the right configuration settings for your cables in the openocd installation folder (usually /usr/share/openocd/scripts/interface/ftdi for FTDI based jtag cables like the digilent ones). The configuration file format is quite well documented in the online documentation of OpenOCD.   Alternatively to copy pasting the content of the interface configuration content into the pulpissimo openocd config file, OpenOCD since a couple of versions also has a mechanism to include to config file without the need for copy pasting ([see here](https://openocd.org/doc/html/Debug-Adapter-Configuration.html). In order to stay backwards compatible with older OpenOCD versions on some of our systems we however did not adapt the syntax yet. 

For what concerns your original comment on the clock gate mapping procedure for our FPGAs:
@bluewww summarized it pretty well. As a research group we do not have the capacity to do in-depth testing of every FPGA port (we do not even have all of them available to test). Also PULPissimo is first and foremost an MCU intended to be implemented as an ASIC. The FPGA port serves just the means to emulate the RTL pretty much as is and is not optimized for speed nor area efficiency on FPGAs. That being said, any form of PR that could improve things is more than welcome.

alruppin, 2023-01-16:
Thank you for your prompt response.
Do you support Vivado simulator? It might not implement all SystemVerilog features, but it is free, and it works under linux and windows.
Supporting it could be VERY nice. ☺
Anyways, I’ve attempted to compile part of model, but got an error message:
ERROR: [VRFC 10-2649] an enum variable may only be assigned the same enum typed variable or one of its values [/raid/algrin/pulpissimo/pulpissimo-master/.bender/git/checkouts/pulp_soc-18e4ef8f70609628/rtl/pulp_soc/soc_interconnect.sv:277]
ERROR: [VRFC 10-2865] module 'soc_interconnect' ignored due to previous errors [/raid/algrin/pulpissimo/pulpissimo-master/.bender/git/checkouts/pulp_soc-18e4ef8f70609628/rtl/pulp_soc/soc_interconnect.sv:24]
Any option you can look at it?
Best regards,
Thanks a lot.

bluewww, 2023-01-16:
Getting our codebase to run on `xsim` (vivado simulator) will be **a lot** of work since we use SystemVerilog language features that are not supported by it.

We intend to release support for verilator in the near future but it is not there yet.

gyx3598, 2023-02-13:
How do you get the testcase under the tests directory? I execute "./update tests". It shows that I have no permission and need a git password. Can you help me?

gyx3598, 2023-02-13:
[image]