pulp-platform/pulpissimo #381: Latest Pulpissimo RTL doesn't fit the nexysA7-100T FPGA

Ahmed1Zaky, 2023-01-05:
2. REPORT DETAILS
-----------------
UTLZ-1#1 Error
Resource utilization  - PBlock:ROOT
RAMB18 and RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 288 of such cell types but only 270 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)ROOT
Related violations: <none>

UTLZ-1#2 Error
Resource utilization  - PBlock:ROOT
RAMB36/FIFO over-utilized in Top Level Design (This design requires more RAMB36/FIFO cells than are available in the target device. This design requires 144 of such cell types but only 135 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)ROOT
Related violations: <none>

UTLZ-1#3 Error
Resource utilization  - PBlock:ROOT
RAMB36E1 over-utilized in Top Level Design (This design requires more RAMB36E1 cells than are available in the target device. This design requires 144 of such cell types but only 135 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.)ROOT
Related violations: <none>

meggiman, 2023-01-16:
Hi @Ahmed1Zaky ,
Could you share the detailed area report you got from the system. I know that recently some design changes to the system were introduced that slightly increased the area utilization of PULPissimo. Maybe we can spot some things MCU features to disable in order to make the design fit your FPGA once again.
Best,
Manuel

Ahmed1Zaky, 2023-01-31:
Hi @meggiman, 

Thanks for your reply. What I have done is that I reduced the size of the L2 a little bit to be able to proceed with the implementation. 

Thanks!

gyx3598, 2023-02-13:
[image]

gyx3598, 2023-02-13:
How do you get the testcase under the tests directory? I execute "./update tests". It shows that I have no permission and need a git password. Can you help me?