pulp-platform/pulpissimo #382: PULP_LDFLAGS += -nostartfiles -nostdlib

aledolme, 2023-01-16:
Goodmorning. 

I'm trying to compile a c program which needs function from stdlib. 
When compiling on Pulpissimo, I obtain the following error:
 **undefined reference to `malloc'
 undefined reference to `free'**

However. when removing  -### nostdlib from PULP_LDFLAGS , it is still not working, and report another error that wasn't present before, related to the toolchain.
**../riscv-pulp/bin/../lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/rv32imcxgap9/ilp32/libgloss.a(syscalls.o):(.sdata+0x0): undefined reference to `_end'**

Any idea of how to solve it? I should not substitute malloc and free function, because I need the c-code to be the same as the one present in https://github.com/PQClean/PQClean for simulation comparison. 

Thanks you

bluewww, 2023-01-16:
You seem to be using gcc 7.1.1 unfortunately I don't have experience with that gcc version

Anyway, normally this means you need to link to newlib which is a minimal libc implementation that should contain these functions (if you are using pulp-runtime)

aledolme, 2023-01-16:
Okay, thanks you! Do you know where and how should I link it?

gyx3598, 2023-02-13:
How do you get the testcase under the tests directory? I execute "./update tests". It shows that I have no permission and need a git password. Can you help me?

gyx3598, 2023-02-13:
[image]

Lorenzoii, 2023-02-13:
I got the same problem ! When I run the simulation, it tells me that the relevant dependent files are missing, and these file paths are in the test folder。I would appreciate your reply！

aledolme, 2023-02-13:
Hi! I cannot remove the two flags I mentioned because the PULP runtime is not compliant with the standard C library. In my case, since the only two functions needed were malloc/free, I replace them with these functions:
void *pi_l2_malloc(int size); 
void pi_l2_free(void *_chunk, int size);
and then it works properly.

gyx3598, 2023-02-13:
> 
How do you get the testcase under the tests directory? I execute "./update tests". It shows that I have no permission and need a git password. Can you help me?

aledolme, 2023-02-13:
I am not using these tests. I am testing some algorithm design by me, and run them on pulpissimo exploiting pulp-runtime, as shown in some example in pulp-platform/sw repository.