pulp-platform/pulpissimo #384: There was an error in the compilation of pulp-runtime examples. Is this the problem with the gcc compiler? Or is there a problem with my Centos7 environment?

gyx3598, 2023-02-20:
There was an error in the compilation of pulp-runtime examples. Is this the problem with the gcc compiler? Or is there a problem with my Centos7 environment?

[image]

bluewww, 2023-02-20:
Please post the compiler version and give a link to the program you are trying to compile.

gyx3598, 2023-02-21:
> 请发布编译器版本并提供指向您尝试编译的程序的链接。

The compiler uses riscv-gnu-toolchain release 2.5.0, and the code is release v7.0.0

gyx3598, 2023-02-21:
> > 请发布编译器版本并提供指向您尝试编译的程序的链接。
> 
> The compiler uses riscv-gnu-toolchain release 2.5.0, and the code is release v7.0.0

[image]

bluewww, 2023-02-21:
Please use the lastest commit from the master branch for pulpissimo. Also I was asking about the *software* you are trying to compile.

meggiman, 2023-02-22:
What software? Please answer the questions asked if you want help. It is impossible for us to know why compilation fails if we don't even know what you are trying to compile. We have no clue what repository and what piece of C-code you are referring to when you mention the "main branch code".

gyx3598, 2023-02-26:
Sorry, I didn't make it clear. Under the main branch, enter the sim directory and execute “make build” and “make simc” respectively. You will find that the PC pointer is dead at 0x00000000: address. As shown below:

[image]

closed 2023-02-27 by gyx3598