pulp-platform/pulpissimo #386: UART module does not work when flash.bin generated from qspi_stim.slm by plp_mkflash.py is used.

yadibelli, 2023-03-07:
Hello All,

We can successfully run "Hello World" test app boots from spi flash memory on RTL simulation mode. We can see UART messages on simulation environment's message terminal. However, we are not able to completely run same app on FPGA (Genesys2) board. It can fetch the instructions and write on GPIO pins (in our case, they point out LEDs). But, unfortunately, we can not read UART port correctly. We can not see any meaningful characters. 

Environment Settings:
flash.bin is generated by plp_mkflash.py from qspi_stim.slm file using following script.
plp_mkflash --flash-boot-binary=build/vectors/test --stimuli=qspi_stim.slm --flash-type=spi --qpi --archi=pulpissimo --raw=flash.bin

ARCHI_PLATFORM_FPGA define is opened in /kernel/chips/pulpissimo/soc.c file. Therefore, following code lines were activated. 
  pos_freq_domains[PI_FREQ_DOMAIN_FC] = ARCHI_FPGA_SOC_FREQUENCY;
  pos_freq_domains[PI_FREQ_DOMAIN_PERIPH] = ARCHI_FPGA_PER_FREQUENCY; 
where 
#define ARCHI_FPGA_PER_FREQUENCY 10000000
#define ARCHI_FPGA_SOC_FREQUENCY 20000000 
code lines were set on /include/archi/chips/pulpissimo/properties.h.

CONFIG_IO_UART=1 and -gBAUDRATE=115200 parameters are also set as is. In spite of all these efforts, we could not get correct uart messages. Where would we make a mistake? We do not know if there is any other file we should touch and change. 

Any help will make us appreciated. Thanks for all.

HaiAnh243, 2023-07-21:
I have a same problem. You can fix this?

micprog, 2023-07-21:
Have you tried compiling the software with `io=uart` added to the make command? I believe the pulp-runtime and pulp-sdk are set up to use the simulation print in hardware by default, but should be switched to uart with this variable.

HaiAnh243, 2023-07-24:
Thanks for your reply. I resolved my problem by using another config file, which used for fpga. I think it fix the same problem like you

closed 2023-07-24 by bluewww