pulp-platform/pulpissimo #387: OpenOCD Error: Debug Module did not become active. dmcontrol=0x0

ivanhira, 2023-03-07:
Hi,

I'm trying to test the SoC on a Nexys Video FPGA. When I try to run OpenOCD after loading the bitstream to the FPGA, the following error occurs:

```
ivantaka@ivantaka-pc:~$ sudo $OPENOCD/bin/openocd -f pulpissimo-caninos/fpga/pulpissimo-nexys_video/openocd-nexys_video.cfg -d
Open On-Chip Debugger 0.10.0+dev-00615-g2b6754e75 (2023-03-07-08:06)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 1 command.c:544 command_print(): debug_level: 3
Debug: 14 1 options.c:184 add_default_dirs(): bindir=/home/ivantaka/pulpissimo-caninos/pulp-sdk/pkg/openocd/1.0/bin
Debug: 15 1 options.c:185 add_default_dirs(): pkgdatadir=/home/ivantaka/pulpissimo-caninos/pulp-sdk/pkg/openocd/1.0/share/openocd
Debug: 16 1 options.c:186 add_default_dirs(): exepath=/home/ivantaka/pulpissimo-caninos/pulp-sdk/pkg/openocd/1.0/bin
Debug: 17 1 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 18 1 configuration.c:42 add_script_search_dir(): adding /root/.openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/ivantaka/pulpissimo-caninos/pulp-sdk/pkg/openocd/1.0/bin/../share/openocd/site
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/ivantaka/pulpissimo-caninos/pulp-sdk/pkg/openocd/1.0/bin/../share/openocd/scripts
Debug: 21 1 configuration.c:97 find_file(): found pulpissimo-caninos/fpga/pulpissimo-nexys_video/openocd-nexys_video.cfg
Debug: 22 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_adapter_khz 1000
Debug: 23 1 command.c:143 script_debug(): command - adapter_khz ocd_adapter_khz 1000
Debug: 25 1 core.c:1645 jtag_config_khz(): handle jtag khz
Debug: 26 1 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 27 1 core.c:1612 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 28 1 command.c:544 command_print(): adapter speed: 1000 kHz
Debug: 29 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface ftdi
Debug: 30 1 command.c:143 script_debug(): command - interface ocd_interface ftdi
Debug: 32 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_device_desc'...
Debug: 33 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_serial'...
Debug: 34 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_channel'...
Debug: 35 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_init'...
Debug: 36 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_signal'...
Debug: 37 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_set_signal'...
Debug: 38 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_get_signal'...
Debug: 39 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_vid_pid'...
Debug: 40 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_tdo_sample_edge'...
Debug: 41 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_oscan1_mode'...
Debug: 42 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_vid_pid 0x0403 0x6010
Debug: 43 1 command.c:143 script_debug(): command - ftdi_vid_pid ocd_ftdi_vid_pid 0x0403 0x6010
Debug: 45 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_channel 0
Debug: 46 1 command.c:143 script_debug(): command - ftdi_channel ocd_ftdi_channel 0
Debug: 48 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_init 0x0018 0x001b
Debug: 49 1 command.c:143 script_debug(): command - ftdi_layout_init ocd_ftdi_layout_init 0x0018 0x001b
Debug: 51 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_ftdi_layout_signal nTRST -ndata 0x0010
Debug: 52 2 command.c:143 script_debug(): command - ftdi_layout_signal ocd_ftdi_layout_signal nTRST -ndata 0x0010
Debug: 54 2 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_transport select
Debug: 55 2 command.c:143 script_debug(): command - ocd_transport ocd_transport select
Info : 56 2 transport.c:286 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 57 2 command.c:364 register_command_handler(): registering 'ocd_jtag_flush_queue_sleep'...
Debug: 58 2 command.c:364 register_command_handler(): registering 'ocd_jtag_rclk'...
Debug: 59 2 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_delay'...
Debug: 60 2 command.c:364 register_command_handler(): registering 'ocd_jtag_ntrst_assert_width'...
Debug: 61 2 command.c:364 register_command_handler(): registering 'ocd_scan_chain'...
Debug: 62 2 command.c:364 register_command_handler(): registering 'ocd_jtag_reset'...
Debug: 63 2 command.c:364 register_command_handler(): registering 'ocd_runtest'...
Debug: 64 2 command.c:364 register_command_handler(): registering 'ocd_irscan'...
Debug: 65 2 command.c:364 register_command_handler(): registering 'ocd_verify_ircapture'...
Debug: 66 2 command.c:364 register_command_handler(): registering 'ocd_verify_jtag'...
Debug: 67 2 command.c:364 register_command_handler(): registering 'ocd_tms_sequence'...
Debug: 68 2 command.c:364 register_command_handler(): registering 'ocd_wait_srst_deassert'...
Debug: 69 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 70 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 71 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 72 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 73 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 74 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 75 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 76 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 77 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
Debug: 78 2 command.c:364 register_command_handler(): registering 'ocd_jtag'...
[... 255 lines cut ...]
Debug: 347 66 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_pld init
Debug: 348 66 command.c:143 script_debug(): command - ocd_pld ocd_pld init
Debug: 350 66 pld.c:205 handle_pld_init_command(): Initializing PLDs...
Debug: 351 66 gdb_server.c:3425 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 352 66 server.c:311 add_service(): Listening on port 3333 for gdb connections
Debug: 353 66 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_halt
Debug: 354 66 command.c:143 script_debug(): command - halt ocd_halt
Debug: 356 66 target.c:3059 handle_halt_command(): -
Error: 357 67 target.c:573 target_halt(): Target not examined yet
Debug: 358 67 command.c:651 run_command(): Command 'halt' failed with error code -4
User : 359 67 command.c:715 command_run_line(): 
Debug: 360 67 riscv.c:411 riscv_deinit_target(): riscv_deinit_target()
Debug: 361 67 riscv-013.c:1423 deinit_target(): riscv_deinit_target()
Debug: 362 67 target.c:1959 target_free_all_working_areas_restore(): freeing all working areas
```

I guess I should mention that I'm using Debian 10 VM on Virtualbox, and the USB seems to work just fine. I did load the bitstream using Vivado from the VM to the FPGA, and the device shows up when running `lsusb`.

The RTL is modified: I removed the FLLs and created PADs for external clock signals, because we didn't have FLL cells to synthesize it. I had to modify the kernel to not initialize the flls, but in the simulation it worked great. I don't think this is the reason it fails, because there is an FPGA FLL that is instantiated when building for FPGAs. But that is the only change in the RTL made.

Also, in the `pulp_soc.sv` file, the `dm_top` module has the port `dmactive_o` connected to nothing. Is that correct? There's a comment  that says "active debug session" that I don't know what it means.

What could be the problem and solution to this?

Thanks in advance.

gyx3598, 2023-03-21:
Can you describe your debugging process? I can debug normally here. I haven't encountered anything like you

ivanhira, 2023-03-21:
Hi, sorry for the late reply.

It was actually a change in the RTL where a module soc_interconnect_wrap was running with peripheral clock instead of the core clock. It was committed by accident by someone, and it wasn't detected in the simulation because they were the same frequency.

Thanks for helping.

prajwalkashyap, 2023-10-11:
Hi @ivanhira I am facing the same issue. How did you fix the error? In the RTL, I can see that soc_interconnect_wrap is not running with the peripheral clock.

l1onog, 2024-04-17:
Hi @ivanhira, @prajwalkashyap  I am facing the same issue. How did you fix the error?