pulp-platform/pulpissimo #398: latest SW support for pulpissimo

akifejaz, 2023-10-13:
Hi, I wanted to ask what is the latest SW support for pulpissimo that is compatible? I've been looking into pulp-runtime and pulp-rt but while configuring having issues running the peripherals test. 
Firstly, I'm able to configure the pulp-runtime, and also able to run the basic tests of which uses stdlib/stdio etc functions.
But when it comes to peripherals testing like i2c, spi it doesn't work. I've tried to add the driver functions from pulp-rt into pulp-runtime but it gives some missing's or mismatched functions issues, that seems hard to resolve.

So, in this regard what is the lasted SW support available that after I configure can test everything, main goal is to write/run test for all peripherals of pulpissimo. 

Suggestions and comments are highly appreciated. Please let me know if I'm following the wrong path or what?

Million thanks!

akifejaz, 2023-10-13:
Can you please have a look at this? your help is much appreciated. 
@bluewww 
@meggiman

bluewww, 2023-10-14:
Unfortunately we don't have the bandwidth to add better software support right now. You can try out  https://github.com/pulp-platform/pulp-freertos which has better driver support

bluewww, 2023-11-08:
If you find a bug please report it in the freertos and if possible propose a fix