pulp-platform/pulpissimo #399: Error connect openocd to debug pulpissimo

namnguyen269, 2023-10-24:
i am implementing Pulissimo project SoC on FPGA. I use Zedboard. I got an error when trying to use openocd to conig jtag for debugging. Please follow the instructions in the readme [https://github.com/pulp-platform/pulpiss...nd-openocd](https://github.com/pulp-platform/pulpissimo/tree/master#gdb-and-openocd)
I git clone and execute the post-build commands 
$ git clone https://github.com/pulp-platform/riscv-openocd
$ ./bootstrap
$ ./config --enable-ftdi
$ ./make
$ ./make install

I then use the command to connect core riscv to the server
$ openocd -f openocd -zedboard-ftdi2232.cfg
I received the following error code:
$openocd -f openocd-zedboard-ftdi2232.cfg 
Open On-Chip Debugger 0.10.0+dev-00830-ga88cc98a0 (2023-10-24-16:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Warn : There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined

Then I switched to another FPGA. specifically the kintex kc705. I received the following error again:
$ openocd -f openocd-kc705.cfg 
Open On-Chip Debugger 0.10.0+dev-00830-ga88cc98a0 (2023-10-24-16:32)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
openocd-kc705.cfg:21: Error: invalid command name "ftdi"
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "openocd-kc705.cfg", line 21

How to configure openocd debugging pulpissimo when developed on FPGA. And I want to develop Pulissimo down to Kintex 7 KC705 FPGA. Can I do it? I'm having trouble

namnguyen269, 2023-10-24:
@bluewww @christian-lanius @meggiman  please help me:((

1yacht, 2025-04-11:
Hello, I have exactly the same problem. Did you find a solution ?