pulp-platform/pulpissimo #400: problem with vivado for windows

RQiand, 2023-11-04:
can I deploy the pulp with the vivado for windows?

bluewww, 2023-11-04:
Windows is not supported as development platform. You would have to try yourself.

kasun-buddhi, 2023-11-08:
Not only Windows but also it is not supported vivado. Am i right bluewww?

RQiand, 2023-11-08:
but the pulp-master repositiory have the fpga folder, Doesn't that mean I can use vivado to deploy pulp to the board？

bluewww, 2023-11-08:
I'm not sure what you are trying to ask. Vivado on linux for deploying on FPGA boards is supported (as you said look into the `fpga`) directory

RQiand, 2023-11-09:
I see. Thank you very much. But here i have one more question: when i try to run the pulp :make checkout, it failed
the information shows as below:
ruirui@ruirui-virtual-machine:~/prj/pulp$ make checkout
make scripts
make[1]: Entering directory '/home/ruirui/prj/pulp'
echo 'set ROOT [file normalize [file dirname [info script]]/..]' > sim/compile.tcl
./bender script vsim \
	--vlog-arg="-suppress 2583 -suppress 13314 \"+incdir+\$ROOT/rtl/includes\"" --vcom-arg="" \
	-t rtl -t test \
	| grep -v "set ROOT" >> sim/compile.tcl
make[1]: *** [Makefile:64: scripts-bender-vsim] Error 1
make[1]: Leaving directory '/home/ruirui/prj/pulp'
make: *** [Makefile:49: checkout] Error 2

i need some help
best wishes

bluewww, 2023-11-09:
I can't tell from the error message unfortunately. Does maybe `./bender` not exist at the location of indicated by the Makefile?

kasunb-accelr, 2023-11-13:
> ivado on linux for deploying on FPGA boards is supported (as you s

I mean simulation on vivado.

kasun-buddhi, 2023-11-13:
> but the pulp-master repositiory have the fpga folder, Doesn't that mean I can use vivado to deploy pulp to the board？

I am not sure. but for simulation, It was mentioned questa. I do not know it is work for vivado

kasun-buddhi, 2023-11-13:
> I'm not sure what you are trying to ask. Vivado on linux for deploying on FPGA boards is supported (as you said look into the `fpga`) directory

how about simulation?

RQiand, 2023-11-14:
> I can't tell from the error message unfortunately. Does maybe `./bender` not exist at the location of indicated by the Makefile?

Thanks a lot. One more question still bothers me: I wanna deploy the pulp cheaply, so Are there any cheap and viable boards to recommend, or how do I judge a board capable for deploying pulp？