pulp-platform/pulpissimo #409: No output on minicom on running hello application for zedboard fpga

zealshah29, 2024-03-26:
<h3 dir="auto">JTAG and UART Connections:</h3>

JTAG Signal | PMOD Pin
-- | --
TMS | JA1
TDI | JA2
TDO | JA3
TCK | JA4
GND | JA5
VCC (trgt) | JA6

UART Signal 	| PMOD Pin
-- | --
RXD | JA7
TXD | JA8
GND | JA11

C Program:
```
#include <stdio.h>

int __rt_fpga_fc_frequency = 20000000; // e.g. 20000000 for 20MHz;
int __rt_fpga_periph_frequency = 10000000; // e.g. 10000000 for 10MHz;
int main()
{
  printf("Hello !\n");

  return 0;
}
```

Compiling test.c file in hello with updated frequencies:
```
c2sd2@c2sd2-HP-Z440:~/pulpissimo/pulp-rt-examples/hello$ make clean all io=UART
rm -rf /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/test /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  
plpflags gen --input=pulpissimo@config_file=fpgas/pulpissimo/genesys2.json  --config=platform=fpga  --config=**/rt/iodev=UART --output-dir=/home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo --makefile=/home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/config.mk    --app=test 
plpconf --input=pulpissimo@config_file=fpgas/pulpissimo/genesys2.json  --config=platform=fpga  --config=**/rt/iodev=UART --output=/home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/config.json 
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c test.c -o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_conf.c -o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o
/opt/riscv/bin/riscv32-unknown-elf-gcc  -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -O3 -g  -fdata-sections -ffunction-sections -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include/io -I/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/include -include /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/fc_config.h    -MMD -MP -c /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.c -o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o
mkdir -p `dirname /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/test`
/opt/riscv/bin/riscv32-unknown-elf-gcc -march=rv32imfcxpulpv2 -mfdiv -D__riscv__ -MMD -MP -o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/test /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc/test.o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_conf.o /home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test/fc//home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/rt_pad_conf.o  -nostartfiles -nostdlib -Wl,--gc-sections -L/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/rules -Tpulpissimo/link.ld -L/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo -L/home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev/install/lib/pulpissimo/pulpissimo_genesys2 -lrt -lrtio -lrt -lgcc
pulp-run --config-file=pulpissimo@config_file=fpgas/pulpissimo/genesys2.json  --config-opt=platform=fpga  --config-opt=**/rt/iodev=UART  --config-opt=**/rt/iodev=UART --dir=/home/c2sd2/pulpissimo/pulp-rt-examples/hello/build/pulpissimo --binary=test/test prepare
```
loading bitstreams into fpga board:
```
c2sd2@c2sd2-HP-Z440:~/pulpissimo/fpga$ make -C pulpissimo-zedboard download
make: Entering directory '/home/c2sd2/pulpissimo/fpga/pulpissimo-zedboard'
Setting environment variables for zedboard board
vivado -mode batch -source tcl/download_bitstream.tcl -tclargs pulpissimo-zedboard.runs/impl_1/xilinx_pulpissimo.bit pulpissimo_zedboard.bit

****** Vivado v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source tcl/download_bitstream.tcl
# open_hw
# connect_hw_server
INFO: [Labtools 27-2285] Connecting to hw_server url TCP:localhost:3121
INFO: [Labtools 27-2222] Launching hw_server...
INFO: [Labtools 27-2221] Launch Output:

****** Xilinx hw_server v2018.3
  **** Build date : Dec  6 2018-23:53:53
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

# open_hw_target [lindex [get_hw_targets -of_objects [get_hw_servers localhost:*]] 0]
INFO: [Labtoolstcl 44-466] Opening hw_target localhost:3121/xilinx_tcf/Digilent/210248AA59CF
# set device [lindex [get_hw_devices] 1]
[... 112 lines cut ...]
Breakpoint 1 at 0x1c008336: file test.c, line 23.
(gdb) c     
Continuing.

Breakpoint 1, main () at test.c:23
23	  printf("Hello !\n");
(gdb)
```

Now I can't see any output on the minicom (3rd terminal)
```
c2sd2@c2sd2-HP-Z440:~/pulpissimo/pulp-rt-examples/hello/build/pulpissimo/test$ sudo minicom
```
```
Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on Aug 13 2017, 15:25:34.
Port /dev/tty8, 15:43:12

Press CTRL-A Z for help on special keys

```
Can anyone please suggest what can be done? 
Thanks!

l1onog, 2024-04-17:
hi, @zealshah29 ,How did you download and compile openocd, when I run the "openocd -f /opt/riscv/pulp/fpga/pulp-zcu102/openocd-zcu102-digilent-jtag-hs2.cfg" command, the error "Debug Module did not become active. dmcontrol=0x0" always appears, which bothers me a lot.

zealshah29, 2024-04-17:
> hi, @zealshah29 ,How did you download and compile openocd, when I run the "openocd -f /opt/riscv/pulp/fpga/pulp-zcu102/openocd-zcu102-digilent-jtag-hs2.cfg" command, the error "Debug Module did not become active. dmcontrol=0x0" always appears, which bothers me a lot.

Hi @l1onog,

So I followed the steps mentioned in the pulpissimo readme file under openocd and gdb section. 
First I installed the following modules:
    • autoconf >= 2.64
    • automake >= 1.14
    • texinfo
    • make
    • libtool
    • pkg-config >= 0.23 (or compatible)
    • libusb-1.0
    • libftdi/ libftdi1-dev
    • libusb-0.1 or libusb-compat-0.1 for some older drivers
• libusb-0.1 or libusb-compat-0.1 for some older drivers
• libusb-0.1 or libusb-compat-0.1 for some older drivers

l1onog, 2024-04-17:
> Hi @l1onog,
>
> So I followed the steps mentioned in the pulpissimo readme file under openocd and gdb section. First I installed the following modules: • autoconf >= 2.64 • automake >= 1.14 • texinfo • [...]

Thanks, May I ask which version of pulp-sdk you are using? and whether you later successfully ran the hello application on the FPGA？

l1onog, 2024-04-17:
> > hi, @zealshah29 ,How did you download and compile openocd, when I run the "openocd -f /opt/riscv/pulp/fpga/pulp-zcu102/openocd-zcu102-digilent-jtag-hs2.cfg" command, the error "Debug Module did not become active. dmcontrol=0x0" always appears, which bothers me a lot.
> 
> Hi @l1onog,
> 
> So I followed the steps mentioned in the pulpissimo readme file under openocd and gdb section. First I installed the following modules: • autoconf >= 2.64 • automake >= 1.14 • texinfo • make • libtool • pkg-config >= 0.23 (or compatible) • libusb-1.0 • libftdi/ libftdi1-dev • libusb-0.1 or libusb-compat-0.1 for some older drivers
> 
> Then, I typed the following in Command Line; source sourceme.sh && ./pulp-tools/bin/plpbuild checkout build --p openocd –-stdout
> 
> I was facing the following errors, I will share them along with their solution.
> 
> ```
> Known issue:
> 1. In function 'memset',
>  inlined from 'mg_gen_ataid' at src/flash/mflash.c:1162:2,
>  inlined from 'mg_storage_config' at src/flash/mflash.c:1174:2:
> /usr/include/x86_64-linux-gnu/bits/string_fortified.h:71:10: error: '__builtin_memset' offset [509, 512] from the object at 'buff' is out of the bounds of referenced subobject 'reserved7' with type 'mg_io_uint8[186]' {aka 'unsigned char[186]'} at offset 322 [-Werror=array-bounds]
> 71 | return __builtin___memset_chk (__dest, __ch, __len, __bos0 (__dest)); | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> Go to directory:
> /home/c2sd2/pulpissimo/pulp-sdk/tools/riscv-openocd/src/flash/mflash.h
> line no 109/110
> 
> mg_io_uint8 reserved7[186];
> 
> to
> 
> mg_io_uint8 reserved7[190];
> 
> cLink: https://pulp-platform.org/community/printthread.php?tid=268
> 2. Configuring PULP SDK at: /home/c2sd2/pulpissimo/pulp-sdk/pkg/sdk/dev
> Traceback (most recent call last):
>   File "./pulp-tools/bin/plpbuild", line 3, in <module>
>     import plptools as plp
>   File "/home/c2sd2/pulpissimo/pulp-sdk/pulp-tools/bin/plptools.py", line 21, in <module>
>     import plpartifactory
>   File "/home/c2sd2/pulpissimo/pulp-sdk/pulp-tools/bin/plpartifactory.py", line 20, in <module>
>     from artifactory import ArtifactoryPath
> ModuleNotFoundError: No module named 'artifactory'
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 	# python3 -mpip install -U dohq-artifactory
> 	clink: https://github.com/devopshq/artifactory/issues/105
> 3. cannot import name 'isoparse' from 'dateutil.parser'
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 	pip install python-dateutil --upgrade
>  https://stackoverflow.com/questions/40809296/upgrade-the-python-package-dateutil-could-not-find-a-version
> _______________________________________________________________
> 4. ModuleNotFoundError: No module named 'twisted'
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 	sudo pip3 install twisted
>  https://stackoverflow.com/questions/47846084/no-module-named-twisted
> _______________________________________________________________
> 5. ModuleNotFoundError: No module named 'openpyxl'
> ModuleNotFoundError: No module named 'xlsxwriter'
> ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> 	sudo apt-get install python3-openpyxl
> python3 -m pip install --user xlsxwriter
> 
>  https://stackoverflow.com/questions/34509198/no-module-named-openpyxl-python-3-4-ubuntu
> ```
> 
> After that I was able to run the following command: `$/home/c2sd2/pulpissimo/pulp-sdk/pkg/openocd/1.0/bin/openocd -f openocd-zedboard-hs2.cfg` Inside `pulpissimo-zedboard` directory

Thank you very much, it will help me a lot!

l1onog, 2024-04-18:
I'm sorry to bother you again, but after I followed your steps, I encountered 

```
"Open On-Chip Debugger 0.10.0+dev-00615-g68f5cfad5 (2024-04-18-09:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
   TapName Enabled IdCode Expected IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 riscv.unknown0 Y 0x00000000 0x10102001 5 0x01 0x03
 1 riscv.cpu Y 0x00000000 0x249511c3 5 0x01 0x03
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.unknown0 tap/device found: 0x5ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x5)
Warn : JTAG tap: riscv.unknown0 UNEXPECTED: 0x5ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x5)
Error: JTAG tap: riscv.unknown0 expected 1 of 1: 0x10102001 (mfg: 0x000 (<invalid>), part: 0x0102, ver: 0x1)
Info : JTAG tap: riscv.cpu tap/device found: 0x24738093 (mfg: 0x049 (Xilinx), part. 0x4738, ver. 0x2)
Warn. JTAG tap: riscv.cpu UNEXPECTED. 0x24738093 (mfg: 0x049 (Xilinx), part. 0x4738, ver. 0x2)
Error. JTAG tap: riscv.cpu expected 1 of 1: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part. 0x4951, ver. 0x2)
Error. Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x08 not 0x01
Warn : Bypassing JTAG setup events due to errors
openocd: src/target/riscv/riscv.c:2366: riscv_xlen_of_hart. Assertion `r->xlen[hartid] ! = -1' failed.
Aborted (core dumped)
"
```
Have you encountered the same problem?

zealshah29, 2024-04-22:
Hello, I am finally able to get the output. Apparently turns out my USB to UART cable was faulty due to which I was not able to get the output.

closed 2024-04-22 by zealshah29

rtspk, 2024-05-09:
@zealshah29 Hello, 

My question is unrelated to your post. I am in the process of implementing RISC-V cores on FPGA. Currently, I am trying to run an RTL simulation. So I wanted to ask, which tool version of Mentor Questa SIM did you use to simulate the design?