pulp-platform/pulpissimo #418: UART not working

RenatoBelmonte, 2024-07-02:
I'm trying to implement a custom hardware accelerator IP in the last version of PULPissimo. 
I can see through gdb that the operation is performed, but when i use `printf()` I can't see anything through UART.  Instead, using the released version it worked.

I've already tried:
- Both SDK and pulp-runtime
- Different programs
- Each ttyUSBx device
- Regenerate the bootrom

The only thing that is printed is 
`booting zforth...
done`

micprog, 2024-07-02:
With the newest update, the I/O multiplexing behavior of PULPissimo has been significantly updated, resulting in required changes to set up the correct I/O mapping that are currently only present in the updated runtime, linked in the repository. Are you using this version of pulp-runtime, or an older one?

RenatoBelmonte, 2024-07-02:
I'm using pulp-sdk v1 to compile application for FPGA such as it is said in the README of pulpissimo (https://github.com/pulp-platform/pulpissimo?tab=readme-ov-file#compiling-applications-for-the-fpga-target)

RenatoBelmonte, 2024-07-11:
I used the least version of pulp-runtime, now UART works properly, but some functions used for manage peripherals don't work. For example, when I set and event with

closed 2024-07-11 by RenatoBelmonte