pulp-platform/pulpissimo #419: Unable to read GPIO as input

Julien2109, 2024-07-23:
Hi,

I'm using the latest stable version of the Pulpissimo (tag v7.0.0: https://github.com/pulp-platform/pulpissimo/tree/v7.0.0), with the Ibex core, running on the Zedboard. 

I'm currently trying to run simple programs for testing different functionalities of the uC, using the simple runtime library. 
I was able to use the UART as well as controlling the GPIO in output mode (blinking the LEDs).

I'm now blocked with the GPIO in input mode. I'm trying to read the values of the switches and push buttons of the Zedboard, but the value stays at 0 for all of them, regardless of the switches position. I tried by configuring and reading the registers manually from the debugger, but again, it's working for the outputs but I'm unable to read the correct value for the inputs. 

I saw some similar cases in the past issues (e.g. https://github.com/pulp-platform/pulpissimo/issues/80#issuecomment-567009855), but I didn't find any answer.  

Is this a known issue or am I missing something ? Thank you in advance for your help !

RenatoBelmonte, 2024-07-23:
Hi, how did you manage the LED blinking?

Julien2109, 2024-08-12:
I wrote the following code:

```c
#include <stdint.h>
#include <stdbool.h>
#include <pulp.h>

void delay_unkn(uint32_t const cycles)
{
    for (uint32_t i = 0; i < cycles; i++)
    {
        asm volatile("nop");
    }
}

int main()
{   
    const uint32_t leds_padfun = 0x00540455;
    const uint32_t leds_mask = 0xE2F;
    // Set the mux to GPIO mode for the leds
    hal_apb_soc_padfun_set(0,leds_padfun);
    // Set the direction of the leds to output
    hal_gpio_set_dir(leds_mask,ARCHI_GPIO_PADDIR_OUT);

    while (true)
    {   
        delay_unkn(500000);
        hal_gpio_set_value(leds_mask,1);

        delay_unkn(500000);
        hal_gpio_set_value(leds_mask,0);
    }

    return 0;
}
```
I did something similar for the inputs, but this is  not working... Same when directly accessing the registers through the debugger.

Julien2109, 2024-08-20:
### **Update and solution**

I found the problem with the GPIOs in input mode. It's actually related to the design of the Zedboard. Specifically, the switches are connected to the FPGA inputs through 10k resistors.  In the PULPissimo design, the GPIO input pads are instantiated with internal pull-down resistors by default. This setup creates a voltage divider when a switch is turned on, preventing the FPGA input from reaching a high enough voltage. As a result, all the inputs remain at a logic "0". Similarly, if we use internal pull-up resistors instead of pull-downs, the voltage divider causes all inputs to stay at a logic "1".

Apparently, the design does not provide pads without any pull-downs or pull-ups. To fix the issue, I added a new module in the file **_pad_functional_xilinx.sv_** (source in the following ip: _tech_cells_generic/src/fpga/pad_functional_xilinx.sv_):

```systemverilog
module pad_functional_no_pull
(
   input  logic             OEN,
   input  logic             I,
   output logic             O,
   input  logic             PEN,
   inout  logic             PAD
);

  (* PULLUP = "NO", PULLDOWN = "NO" *)
  IOBUF iobuf_i (
    .T ( OEN ),
    .I ( I    ),
    .O ( O    ),
    .IO( PAD  )
  );

endmodule
```
Then, I replaced the default pad with pull-down by the new module for the involved signals (related to the switches and push-buttons inputs) in the file **_pad_frame.sv_** (source in _rtl/pulpissimo/pad_frame.sv_) -> lines 178 to 208:
```systemverilog
pad_functional_no_pull padinst_sdio_data0 (.OEN(~oe_sdio_data0_i ), .I(out_sdio_data0_i ), .O(in_sdio_data0_o ), .PAD(pad_sdio_data0 ), .PEN(~pad_cfg_i[22][0]) );
pad_functional_no_pull padinst_sdio_data1 (.OEN(~oe_sdio_data1_i ), .I(out_sdio_data1_i ), .O(in_sdio_data1_o ), .PAD(pad_sdio_data1 ), .PEN(~pad_cfg_i[23][0]) );
pad_functional_no_pull padinst_sdio_data2 (.OEN(~oe_sdio_data2_i ), .I(out_sdio_data2_i ), .O(in_sdio_data2_o ), .PAD(pad_sdio_data2 ), .PEN(~pad_cfg_i[24][0]) );
pad_functional_no_pull padinst_sdio_data3 (.OEN(~oe_sdio_data3_i ), .I(out_sdio_data3_i ), .O(in_sdio_data3_o ), .PAD(pad_sdio_data3 ), .PEN(~pad_cfg_i[25][0]) );
pad_functional_pd padinst_sdio_clk   (.OEN(~oe_sdio_clk_i  ), .I(out_sdio_clk_i  ), .O(in_sdio_clk_o  ), .PAD(pad_sdio_clk  ), .PEN(~pad_cfg_i[20][0]) );
pad_functional_pd padinst_sdio_cmd   (.OEN(~oe_sdio_cmd_i  ), .I(out_sdio_cmd_i  ), .O(in_sdio_cmd_o  ), .PAD(pad_sdio_cmd  ), .PEN(~pad_cfg_i[21][0]) );
pad_functional_pd padinst_spim_sck   (.OEN(~oe_spim_sck_i  ), .I(out_spim_sck_i  ), .O(in_spim_sck_o  ), .PAD(pad_spim_sck  ), .PEN(~pad_cfg_i[6][0] ) );
...
pad_functional_pd padinst_cam_pclk   (.OEN(~oe_cam_pclk_i  ), .I(out_cam_pclk_i  ), .O(in_cam_pclk_o  ), .PAD(pad_cam_pclk  ), .PEN(~pad_cfg_i[9][0] ) );
pad_functional_pd padinst_cam_hsync  (.OEN(~oe_cam_hsync_i ), .I(out_cam_hsync_i ), .O(in_cam_hsync_o ), .PAD(pad_cam_hsync ), .PEN(~pad_cfg_i[10][0]) );
pad_functional_pd padinst_cam_data0  (.OEN(~oe_cam_data0_i ), .I(out_cam_data0_i ), .O(in_cam_data0_o ), .PAD(pad_cam_data0 ), .PEN(~pad_cfg_i[11][0]) );
pad_functional_no_pull padinst_cam_data1  (.OEN(~oe_cam_data1_i ), .I(out_cam_data1_i ), .O(in_cam_data1_o ), .PAD(pad_cam_data1 ), .PEN(~pad_cfg_i[12][0]) );
pad_functional_no_pull padinst_cam_data2  (.OEN(~oe_cam_data2_i ), .I(out_cam_data2_i ), .O(in_cam_data2_o ), .PAD(pad_cam_data2 ), .PEN(~pad_cfg_i[13][0]) );
pad_functional_no_pull padinst_cam_data3  (.OEN(~oe_cam_data3_i ), .I(out_cam_data3_i ), .O(in_cam_data3_o ), .PAD(pad_cam_data3 ), .PEN(~pad_cfg_i[14][0]) );
pad_functional_no_pull padinst_cam_data4  (.OEN(~oe_cam_data4_i ), .I(out_cam_data4_i ), .O(in_cam_data4_o ), .PAD(pad_cam_data4 ), .PEN(~pad_cfg_i[15][0]) );
pad_functional_no_pull padinst_cam_data5  (.OEN(~oe_cam_data5_i ), .I(out_cam_data5_i ), .O(in_cam_data5_o ), .PAD(pad_cam_data5 ), .PEN(~pad_cfg_i[16][0]) );
pad_functional_no_pull padinst_cam_data6  (.OEN(~oe_cam_data6_i ), .I(out_cam_data6_i ), .O(in_cam_data6_o ), .PAD(pad_cam_data6 ), .PEN(~pad_cfg_i[17][0]) );
pad_functional_no_pull padinst_cam_data7  (.OEN(~oe_cam_data7_i ), .I(out_cam_data7_i ), .O(in_cam_data7_o ), .PAD(pad_cam_data7 ), .PEN(~pad_cfg_i[18][0]) );
```
After that, I've been able to read the correct values in the register when playing with the switches and buttons on the Zedboard. Hopefully this may help someone else in the future...

closed 2024-08-20 by Julien2109

RenatoBelmonte, 2024-08-27:
Thank you for sharing