pulp-platform/pulpissimo #429: Solutions to exercises of "A Deep Dive into HW/SW Development with PULP" training

julienissuts, 2024-10-07:
Hello,

I have been going through the "A Deep Dive into HW/SW Development with PULP" training, but I was unable to locate the templates for the exercises or their corresponding solutions. Could you kindly provide these materials or direct me to where they can be found?

Thank you and best regards,
Julien

RenatoBelmonte, 2024-10-07:
Are you referring to this [https://github.com/pulp-platform/pulp-training](url) ?  In addition, Manuel Eggimann provided an exhaustive solution for the last example here 
[https://github.com/meggiman/wide_alu_ip](url)

julienissuts, 2024-10-07:
Thanks for the fast response. I was referring to the video: [https://www.youtube.com/watch?v=B7BtaYh3VqI&t=15359s](https://www.youtube.com/watch?v=B7BtaYh3VqI&t=15359s)

There are some exercises about the reggen tool, the boot sequence etc. and i couldn't find the templates or solutions for them. 

The link you sent isn't exactly that but seems also very useful thanks.

julienissuts, 2024-10-07:
Okay I found the repository of all the examples mentioned in the training:
[A Deep Dive into HW/SW Development with PULP: Exercises](https://github.com/pulp-training/sw)

closed 2024-10-07 by julienissuts