pulp-platform/pulpissimo #432: Error message whilst running the 'hello' example

LiamSkirrow, 2024-10-28:
Hi there! I'm attempting to run the hello example obtained from [here](https://github.com/pulp-platform/pulp-runtime-examples), but am running into the following error message:
```
liam-skirrow:~/Documents/code/pulpissimo/sw/pulp-runtime/hello$ make all
Setting platform to rtl
CC  test.c
mkdir -p `dirname /home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/hello/build/test/test.o`
/pulp-gcc-2.5.0//bin/riscv32-unknown-elf-gcc  -c test.c -o /home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/hello/build/test/test.o -MMD -MP -march=rv32imcxgap9 -O3 -g -D__riscv__ -fdata-sections -ffunction-sections -include chips/pulp/config.h -I/home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/include/chips/pulp  -fno-jump-tables -fno-tree-loop-distribute-patterns -D__PLATFORM__=ARCHI_PLATFORM_RTL -DCONFIG_IO_UART=0 -DCONFIG_IO_UART_BAUDRATE=115200 -DCONFIG_IO_UART_ITF=0 -I/home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/lib/libc/minimal/include -I/home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/include -I/home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/kernel 
internal error: can't hash `pv.add.h.div2': exists
Assembler messages:
Fatal error: Broken assembler.  No assembly attempted.
make: *** [/home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/rules/pulpos/default_rules.mk:257: /home/liam-skirrow/Documents/code/pulpissimo/sw/pulp-runtime/hello/build/test/test.o] Error 1
```

I have built the [riscv toolchain](https://github.com/pulp-platform/riscv-gnu-toolchain) and have placed the 'hello' example at the path `pulpissimo/sw/pulp-runtime/hello` which I believe is correct, but would like to have that confirmed. Any pointers would be appreciated since I'm pretty stuck on this at the moment.

I'm running on Ubuntu 24.04.1 LTS in case that's relevant.

Thanks!

LiamSkirrow, 2024-10-28:
I wasn't totally sure where to place the `pulp-runtime-examples/hello` directory so I experimented with placing it at other locations, for example at the top-level project directory `pulpissimo/` but this made no difference and I still see the same `Fatal error: Broken assembler` message. This makes it seems like an issue with the compiler perhaps?

dave-roo, 2024-10-29:
I'm also having this problem.

bluewww, 2024-10-31:
I just tried to reproduce your problem on the master branch with the last commit but I can't. The hello example just compiles fine. This is the compiler version I used

```
hello$ riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (2021-10-30 PULP GCC v2.5.0) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

LiamSkirrow, 2024-10-31:
Ok thanks for checking, when checking my compiler version it turns out I'm using the same version...

```~/Documents/code/pulpissimo((HEAD detached at v7.0.0))$ riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (2021-10-30 PULP GCC v2.5.0) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Same as you. 

I'm intending to use the `v7.0.0` release anyway, so this isn't too much of a blocking issue for me, but I thought it'd be good to flag here anyway for others.

dave-roo, 2024-11-05:
I've used the latest run-time. 
https://github.com/pulp-platform/pulp-runtime

I'm now installing Ubuntu 16.04 as this maybe the issue.